contiki-os/contiki #1273: sht11_temp() fix for Z1 platform not working

kriegalex, 2015-09-23:
The fix located at

https://github.com/contiki-os/contiki/commit/c5e9d688a5a6589669e5e4553a3d2c35df264dc6

and discussed here

http://sourceforge.net/p/contiki/mailman/message/29682840/

does't seem to work anymore. Another person has reported the problem on stackoverflow : 

http://stackoverflow.com/questions/29338515/wrong-temperature-humidity-results-using-sht11-zig001

Can someone validate that there is a problem (again) ?

alignan, 2015-09-23:
The problem was due to the SHT11 not being a proper I2C sensor.  I would rather recommend moving to the SHT25 or alike.  I could test this, however this will not happen in 2-3 weeks.

kriegalex, 2015-09-23:
Hi, thanks for the quick answer. I've been given this project, so in the current contiki version, there was no sht25, so with the help of your reply I was able to retrieve the missing files from github.

Unfortunately, the new test-sht25 gives me negative values (read incorrect), but this is another topic. Thanks anyway, I'll update the stack overflow and see if the person still needs the information.

alignan, 2015-09-23:
Mmm OK, let me give it a try tomorrow with the SHT25 (I don't have any at home) and get back to you, thanks!

alignan, 2015-11-23:
Hello!

Could you also add this in your `project-conf.h` or `Makefile`:

`#define I2C_CONF_RX_WITH_INTERRUPT   0`

Seems to work for me in most cases

kriegalex, 2015-11-24:
Hi,

Sorry but I have been confused by the posts on stackoverflow and the mailing list and it made believe that sht11 was indeed supposed to work on Z1. It is true only if you use a zigglet, but the built in captor is in fact using the tmp102.h library and not sht11.h. I didn't know that.

As I was given this project, I took for granted what was written, and seeing those posts on the internet I thought I was not alone. I should have checked if Z1 was really using sht11. I migrated the code to tmp102 and it works now on the Z1.

Sorry if I may have make you lose some time there.

g-oikonomou, 2015-11-24:
Closing this (?)

closed 2015-11-24 by g-oikonomou