contiki-os/contiki #1277: NDP bug

simonduq, 2015-09-24:
The function `uip_ds6_nbr_get_ll` returns `const uip_lladdr_t *`, a pointer to the read-only link-layer address used by `nbr-table` as index.

In `uip-nd6.c`, the return value is cast to non-constant pointer in several locations, with:
`uip_lladdr_t *lladdr = (uip_lladdr_t *)uip_ds6_nbr_get_ll(nbr);`
This is followed by writing to `lladdr`, which will corrupt the `nbr-table` module.

A fix would be to remove the old neighbor and add a new one rather than try to write the address in place.

joakimeriksson, 2015-11-12:
Yes, this is an issue. I think we have just been hit by that bug - it can cause multiple instances of the same lladdr in the nbr-table module which can cause anything including the inability to communicate at all with that neighbor (unrecoverable until global repair in RPL). The only time it is ok to change would be when the lladdr is NULL. But maybe it would be better to not allow for any NULL element in the nbr-module and handle the NULL element separately since it is only one other module that make use of it (uip-nd6). Then there will be no allocations in the nbr module when there is a INCOMPLETE entry installed, but only when it changes to a REACHABLE or STALE state and have both IPv6 address and lladdr.

simonduq, 2015-11-12:
Handling incomplete entries separately would probably be the best I agree.
But in a first time, I think allowing a single NULL entry (like in the current codebase), but making sure to always remove then add but never write in place should fix the bug.
Note: I do not think it is ever ok to write to the lladdr of a table entry from outside of the nbr-table module; This might result in duplicates, even when the address was NULL before or is becoming NULL.

joakimeriksson, 2015-11-12:
Yes totally agree that no other module can write the lladdr. If we keep the NULL entry and want to change the lladdr it can only be made by the module. But I guess the safest way is to remove and add. The only case so far when NULL is used is the ND when Incomplete. Maybe we just should never store an Incomplete entry? E.g. we just add a REACHABLE when we get a NA with is-solicited flag set (then we know that we sent the NS).

simonduq, 2015-11-12:
Both options would work for me (adding only reachable entries OR remove incomplete then add reachable)

joakimeriksson, 2015-11-12:
The nice thing with  not adding incomplete would be that we could slightly simplify nbr-module and avoid throwing out a possibly important neighbour just because we happen to get an incomplete entry.  For any cases where there are no NA coming back we just do not throw anyone else out.

joakimeriksson, 2015-11-19:
Having spent another week on this - I know now that this is a significant problem - but I I do have two different fixes for it. (1) replacing the memcpy that cause multiple entries of the same lladdress. This I can upstream tomorrow and (2) re-implementing how NS/NA is handled for INCOMPLETE entries - this will need some more testing.

These problems cause disconnects and the (1) solution seems to solve it. I have not yet seen any permanent disconnects from a node since I fixed that. Have tested during "millions" of pings and with really challenging configurations to get the bug triggered - but no problem so far.

simonduq, 2015-11-19:
Upstreaming (1) tomorrow sounds great as I'm confident this really fixes the problem. (2) is even nicer but will take more time to nail down.