contiki-os/contiki #1278: Static routes are not possible since neighbors table refactoring

laurentderu, 2015-09-24:
Since the neighbor table refactoring, it is no longer possible to have static IPv6 routes in Contiki. To create a route the nexthop must be already a neighbor of the host and its mac address must be known. This prevent statically adding routes as it either require to also create the neighbor entry for the next hop or trigger a NS to receive its MAC address.
Also, as a route is tightly linked to the next hop presence in the neighbor table, when that entry is removed (for lack of room), the route is deleted. Also if the route is not often used it could get removed if the routing table becomes full.

Static routes are needed on network not using RPL but still require some routing (NDP based WSN or 6LoWPAN-ND based WSN), also they are needed by Border Routers when the LAN has a more complex topology.

In order to bring back support for static routes, we could add back the next hop IP address in the route struct, however this will add 16 bytes per route, which is not ideal for RAM constrained nodes (It could be disabled using a compilation flag). Additional flags would also be needed to differentiate between static routes and dynamic routes.
Another option would be to have static routes having their own struct and management code, only the route selection function would scan both lists to find the suitable next hop. The support for static route could be disabled using a compilation flag; also the support for dynamic routes could be disabled for nodes not using RPL.

I'm leaning towards solution 2 and I will propose a PR if there are no arguments against this solution.

joakimeriksson, 2015-09-25:
Would we not rather try to make this supported by adding and locking the next-hop neighbor needed and install
the route? You will anyway need the neighbor when you send to the nexthop so at runtime it needs to be there
anyway?

The PR I am working with that support end-to-end DAO ACK and handling of DAO NACK also protects routes
from being deleted other than by timeouts.  Combing that with the correct routing would make it easier to get back
support for static routes without adding back 16-bytes into the routing table entries.

Best regards
— Joakim

laurentderu, 2015-09-25:
That would help not bring back the next hop IP address or to have another list of routes struct, but you would still have the burden or resolving the next hop MAC address before being able to add the actual route.
That means when a node with static routes starts up, you will have a bunch of NS/NA messages occurring  to resolve all the MAC addresses of the all the next hop, even if the route are not used immediately. Also, the add_static_route code will be complex as it has to wait the result of the NA before being able to add the route. And finally, if a next hop is not yet reachable, the route will not be created even if at some later time the host becomes reachable.
That's why I'm more in favor or a list of static routes, with their next hop resolved only when needed. This list could be short or even removed altogether on hosts that does not need static routes.

simonduq, 2015-09-25:
Are you in a scenario where you know the next hop link-local IPv6, but not its MAC?
If you have the MAC you can always add the MAC <-> link-local IPv6 mapping statically, before the route.

If not, I agree the current implementation is sub-optimal... even adding the address in the route struct doesn't seem sufficient to me, because you'd need the MAC address anyway at some point to add the neighbor to nbr-table. You probably something with its own management (your solution 2).

As for the locking: neighbors used as nexthop are currently locked, so routes should not be removed for lack of space.

laurentderu, 2015-09-25:
Yes, you can have a route configured without yes knowing it's MAC, either because it's somehow available in your host by configuration or hardcoded, or received via RIO option in RA. (You could also receive a static route in RPL through the RIO option, but without multiple instances support I don't think it's useful). In both case, the MAC address is unknown and can not be retrieved in the even the next hop host is not yet reachable.
We don't need to store the next-hop MAC address or create the next-hop entry in the neighbor table however, in tcpip.c if the next hop is not a known neighbor, the code will send a NS message to resolve its MAC address and store the outgoing packet in the packet queue.
Locking is done by RPL code, if you are in non-RPL mode there is no locking whatsoever.

simonduq, 2015-09-25:
I see. So yes I think you'd need some distinct management for static routes then!

joakimeriksson, 2015-09-25:
Locking should no longer be done by RPL but the routing module itself (recent PR that was pulled).

Best regards,
— Joakim

joakimeriksson, 2016-03-29:
Anyone working on this?

simonduq, 2016-03-29:
not me

laurentderu, 2016-03-29:
I made a static route module as a proof of concept, I can clean it up and propose it as PR so we can discuss over concrete code.

iqbal-h, 2018-06-03:
Anyone still working on this? Static routing with RPL and TSCH?