contiki-os/contiki #1283: Question: cc26xx web demo IPv6 address assignment method

xianlin, 2015-09-28:
This is a question:

I read that 6lowpan ip address assignment has **dynamic, static and hybrid** (refer [here](http://www.embedded.com/electronics-blogs/embedded-cloud-talkers/4236873/How-to-setup-a-6LoWPAN-network)). 

I wonder in the cc26xx-web-demo example, which IP address assignment method is being used? and in which line/part of the code is the setting for this IP assignment. Anyone kind share your know how on this? Thank you.

tonnenpinguin, 2015-09-29:
As far as I know the IP address is copied from cc26xx's info page if you do not change this setting in your project-conf.h. The IP address is set as part of the initialization code in contiki-main.c (check the respective platform folder for that file).
So I guess you can call it static ip address assignment.

xianlin, 2015-09-29:
@tonnenpinguin thank your for sharing. 

I saw the code on [line 130](https://github.com/contiki-os/contiki/blob/master/platform/srf06-cc26xx/contiki-conf.h#L130) but I cannot understand what is the **info page**.

The definition of "info page" i can found is at `\cpu\cc253x\cc253x.h` line 670 

```c
/*---------------------------------------------------------------------------
 * Information Page (Read Only)
 *---------------------------------------------------------------------------*/
SFRX(X_INFOPAGE,  0x7800); /* Start of Information Page */
SFRX(X_IEEE_ADDR, 0x780C); /* Start of unique IEEE Address */
```

And at `\cpu\cc26xx-cc13xx\ieee-addr.h` line 44.

```c
\defgroup cc26xx-ieee-addr CC13xx/CC26xx IEEE Address Control
 *
 * Driver for the retrieval of an IEEE address from flash
 *
 * The user can specify a hardcoded IEEE address through the
 * IEEE_ADDR_CONF_HARDCODED configuration macro.
 *
 * If the user does not hard-code an address, then one will be read from either
 * the primary location (InfoPage) or from the secondary location (on flash).
 *
 * In order to allow the user to easily program nodes with addresses, the
 * secondary location is given priority: If it contains a valid address then
 * it will be chosen in favour of the one on InfoPage.
 *
 * In this context, an address is valid if at least one of the 8 bytes does not
 * equal 0xFF. If all 8 bytes are 0xFF, then the primary location will be used.
 *
 * In all cases, the address is assumed to be written little-endian.
 *
 * Lastly, it is possible to override the 2 LSB's of the address by using the
 * NODE_ID make variable.
```

Where is the above `primary location` on the CC2650 mcu? There is an IPv6 address stored? By who or how to stored before MCU powered? Thank you.

tonnenpinguin, 2015-09-29:
The Info Page is located in the internal flash.
As you found out correctly it can be read using TI's smartRF Flash Programmer 2. There you can also see the absolute flash address that the MAC address can be read from.
For more information about how the IPv6 address can be calculated from the MAC address please refer to [this](http://packetlife.net/blog/2008/aug/4/eui-64-ipv6/) link!

xianlin, 2015-09-30:
@tonnenpinguin , thank you and that solved my problem. 
we can close this issue now.

closed 2015-09-30 by xianlin