contiki-os/contiki #1287: CC2538 Vector Table Offset in nvic_init() at nvic.c

randy-marsh, 2015-09-29:
We don't need to set REG(SCB_VTABLE) to NVIC_CONF_VTABLE_BASE, because the boot ROM will already have set up our interrupt vector table at boot time. The CC2538 sets the vector table offset using the direction assigned in the cca area. If  we made an image in a flash area higher than 0x200000 contiki will crash because nvic_init() moves the vector table again to 0x200000.

bthebaudeau, 2015-09-29:
You can define `NVIC_CONF_VTABLE_BASE` and `NVIC_CONF_VTABLE_OFFSET` to what you need, and use a custom startup-gcc.c. This assignment is here to make it possible to do custom things that may be different from the value in the CCA.

But the current mechanism is indeed not great. It's greatly improved by #1162.

bthebaudeau, 2015-11-23:
Fixed by #1162.

closed 2015-11-23 by bthebaudeau