contiki-os/contiki #1291: contikimac: sub-ghz support (cc1200)

uknoblic, 2015-10-01:
Hi,

we are currently tuning contikimac for the cc1200 driver. Looking at contikimac, it seems that at some places there is missing support for sub-ghz (aka low-speed) devices. For instance MAX_PHASE_STROBE_TIME fixed to 16ms which means that just 1 packet will be transmitted once we are locked which is not appropriate. Any comments an this? Is there any 50kbit radio tested with contikimac?

g-oikonomou, 2015-10-01:
#1160

g-oikonomou, 2015-10-01:
I should say, the only time I experimented with phase optimisation was at 2.4GHz and I didn't manage to make it work, hence `MAX_PHASE_STROBE_TIME` was not included in the PR above. I would start without phase locks to make sure the driver plays nicely before trying to enable them. Further configurability for ContikiMAC may prove necessary as you say.

uknoblic, 2015-10-01:
Thanks George - will keep you updated!

closed 2015-10-13 by uknoblic