contiki-os/contiki #1295: CC26xx-web-demo enable AES encryption in Contiki 3.0

xianlin, 2015-10-02:
**Claim:**

I have tried to send the below email to `contiki-developers@lists.sourceforge.net` but was rejected by the system. (I have subscribed to the mailing list)

**Below is my question:**

Contiki 3.0 has the new security feature as encryption.
http://contiki-os.blogspot.sg/2014/10/a-big-step-for-contiki-built-in.html

How to enable AES encryption for the contiki `CC26xx-web-demo` in the `example/cc26xx` folder?

Or better question is: 
**How to enable the AES encryption for 802.15.4 in contiki for any CPU/Platform?**

Do I need to add extra line into "project.conf" such as:
```c
#define AES 1
```

Thank you

gillesDD, 2015-10-02:
hi,

add   *NETSTACK_LLSEC.bootstrap(**start_network_layer);* in the main
function from contiki/platform/srf06-cc26xx/contik_main.c and define
start_network_layer as a foo function.
See example from sky plateform

then add in your project-conf.h

*#undef NETSTACK_CONF_LLSEC#define NETSTACK_CONF_LLSEC
noncoresec_driver#undef LLSEC802154_CONF_SECURITY_LEVEL#define
LLSEC802154_CONF_SECURITY_LEVEL   YOUR_LEVEL *

also, add noncoresec to be compile in the Makefile.
contiki/platform/srf06-cc26xx/srf06-cc26xx
MODULES += core/net core/net/mac core/net/mac/contikimac core/net/llsec
*core/net/llsec/noncoresec*

gilles

xianlin, 2015-10-02:
@gillesDD thanks and I will give it a try. 

I forgot that I have to enable AES on my **Border Router** as well which is a CC2530 USB dongle connected with Beagle Bone Black. 

I have posted an [issue](https://github.com/cetic/6lbr/issues/105) to ask on [Cetic/6lbr repo](https://github.com/cetic/6lbr). 

Or maybe @gillesDD can help again? Thank you.

xianlin, 2015-10-02:
Based on what I read at the [wiki page](https://github.com/kkrentz/contiki/wiki/LLSEC) of contiki, I cannot use **hardware AES** on CC2650 sensortag yet. Or can I?

software AES will drain my poor sensortag coin cell battery pretty fast. Have to use DC power if security must be implemented. :-(

gillesDD, 2015-10-03:
AES encryption is to be configured from the WEB interface of 6lbr, and it
is running in the Contiki 6lbr beagleboard
You don't have to activate encryption in the cc2531 dongle

also, AES HW API is available from TI lib , but has to be bind to llsec
driver.

I tested this API using cc2538 crypto test example, but SW remains block at
the time key storage is configured.
I ask Texas support, still no answer

xianlin, 2015-10-04:
@gillesDD I **wish** that is true for 6lbr web GUI to include this feature (enable AES key).

Unfortunately there is NO such thing in 6lbr web GUI.

[image]

So still no answer to enable AES in either CC2530 USB or 6lbr on Beagle Bone Black.

gillesDD, 2015-10-04:
You are not on the bleeding edge repo, update your 6lbr and you'll have the
WEB GUI updated for settings

xianlin, 2015-10-05:
@gillesDD Thank you and I have successfully compile and installed the newest code [develop-20150825](https://github.com/cetic/6lbr/releases/tag/develop-20150825).

[image]

It now can show the configuration page for the 802.15.4 security:

[image]

Next I will try to modify and compile the contiki firmware with AES security enabled on sensortag as [the above instructions](https://github.com/contiki-os/contiki/issues/1295#issuecomment-144973317) and test the communications with this new 6lbr border router.

xianlin, 2015-10-05:
The AES encryption code is enabled on sensor tag web demo and here are my code changes, please note the file for the changes are:

`platform/srf06-cc26xx/contiki-main.c`

`platform/srf06-cc26xx/Makefile.srf06-cc26xx`

`examples/cc26xx/cc26xx-web-demo/project-conf.h`

https://github.com/xianlin/contiki/commit/27ed39dca56f7528879c803e0672481f3b9013c8

**Note:**

 - The new 6lbr web GUI PAN ID is shown in decimal format for PAN ID hex format. i.e. If you want to set PAN ID = 0xABCD, please input as `ABCD` but after save and reboot, the web GUI will display a deciaml value of `43981`
 - For **Link Layer Security** details please refer to this [wiki page](https://github.com/kkrentz/contiki/wiki/LLSEC).
 - For AES security level, please refer to the file https://github.com/kkrentz/contiki/blob/master/core/net/mac/ieee802154.h#L95-L104, note that in the webGUI the list sequence will be identical to the above enum sequence, i.e. enum[0]=web_drop_down[first_listed_item].
 - Although the AES is working but it seems sometimes it is very difficult to get to the web server or COAP resources in the `cc26xx-web-demo`, the web browser will show `time out` errors. This will be investigated in another issue.

xianlin, 2015-10-05:
@gillesDD Thank you very much for the help. It is solved :+1:

closed 2015-10-05 by xianlin

bbhagasra1, 2015-11-26:
All the above modification needed to change for slip radio also??

gillesDD, 2015-11-26:
No change on slip radio. Encryption is running on 6lbr itself

bbhagasra1, 2015-11-26:
then what version of wireshark needed for sniffer.

On 11/26/2015 5:02 PM, gillesDD wrote:
> r itself

-- 
Thanks
Bijendra Bhagasra

Email Confidentiality Notice and Disclaimer
This email message contains information which may be confidential and or legally privileged. Unless you are the intended recipient (or have been authorized to receive on behalf of the intended recipient), you may not use, copy or disclose to anyone the message or any information contained in the message or from any attachments that were sent with this email. If you have received this email message in error, please advise the sender by email, and delete the message. Further, this message contains views and comments of the individual sending the message and the company is in no way liable for any issues arising from use from such a message.

----------------------------------------------
Genus Power Infrastructures Ltd, Jaipur, India

bbhagasra1, 2015-12-07:
slip radio not sending any data on rf port when AES enabled in 6lbr.

msolters, 2016-01-22:
Has AES implementation changed since the time of this writing?  Following the above instructions, Contiki fails to compile with the following error:

```
../../../platform/srf06-cc26xx/./contiki-main.c: In function 'main':
../../../platform/srf06-cc26xx/./contiki-main.c:220:17: error: 'const struct llsec_driver' has no member named 'bootstrap'
   NETSTACK_LLSEC.bootstrap(start_network_layer);
                 ^
../../../Makefile.include:240: recipe for target 'obj_srf06-cc26xx/contiki-main.o' failed
make: *** [obj_srf06-cc26xx/contiki-main.o] Error 1

```

msolters, 2016-01-22:
Indeed, there are no LLSEC-related instances of keyword "bootstrap" anywhere in the Contiki repo.