contiki-os/contiki #1298: C26xx BLE stack

spoerk, 2015-10-07:
I am currently starting to write my master thesis about BLE and IoT and use the TI SensorTag as the platform to do so. The result of my thesis should be [IPv6 over BLE](https://tools.ietf.org/html/draft-ietf-6lo-btle-17).

After looking at the current source code of the C26xx drivers, I see that contiki does not have a BLE stack. Is there any development in regards for a BLE stack on contiki, that is not yet included in the master branch?

simonduq, 2015-10-07:
Not that I'm aware of but @EarthLord wrote this
https://github.com/EarthLord/contiki

This is a port to Contiki to the Nordic Semiconductor nrf51822 (uses Nordic's proprietary BLE stack)

tonnenpinguin, 2015-10-16:
@spoerk If you decide to implement your own cc26xx BLE stack I would volunteer to contribute as well!

craigm26, 2015-10-16:
this would be amazing

spoerk, 2015-12-17:
I found a [TI BLE stack](http://www.ti.com/tool/ble-stack) for the cc26xx and RTOS. 
This stack implements all BLE functionality according to the BLE 4.1 specification.

I would look into porting this BLE stack to contiki. The problem is, that the license of this stack is restrictive.
I think the license is too restrictive for including it into contiki.

Has anyone any experience with talking to TI about their source code licenses?

I would port the stack to contiki, if the license was no problem.

License text:
```
/*
  Copyright 2009 - 2015 Texas Instruments Incorporated. All rights reserved.

  IMPORTANT: Your use of this Software is limited to those specific rights
  granted under the terms of a software license agreement between the user
  who downloaded the software, his/her employer (which must be your employer)
  and Texas Instruments Incorporated (the "License").  You may not use this
  Software unless you agree to abide by the terms of the License. The License
  limits your use, and you acknowledge, that the Software may not be modified,
  copied or distributed unless embedded on a Texas Instruments microcontroller
  or used solely and exclusively in conjunction with a Texas Instruments radio
  frequency transceiver, which is integrated into your product.  Other than for
  the foregoing purpose, you may not use, reproduce, copy, prepare derivative
  works of, modify, distribute, perform, display or sell this Software and/or
  its documentation for any purpose.

  YOU FURTHER ACKNOWLEDGE AND AGREE THAT THE SOFTWARE AND DOCUMENTATION ARE
  PROVIDED "AS IS" WITHOUT WARRANTY OF ANY KIND, EITHER EXPRESS OR IMPLIED,
  INCLUDING WITHOUT LIMITATION, ANY WARRANTY OF MERCHANTABILITY, TITLE,
  NON-INFRINGEMENT AND FITNESS FOR A PARTICULAR PURPOSE. IN NO EVENT SHALL
  TEXAS INSTRUMENTS OR ITS LICENSORS BE LIABLE OR OBLIGATED UNDER CONTRACT,
  NEGLIGENCE, STRICT LIABILITY, CONTRIBUTION, BREACH OF WARRANTY, OR OTHER
  LEGAL EQUITABLE THEORY ANY DIRECT OR INDIRECT DAMAGES OR EXPENSES
  INCLUDING BUT NOT LIMITED TO ANY INCIDENTAL, SPECIAL, INDIRECT, PUNITIVE
  OR CONSEQUENTIAL DAMAGES, LOST PROFITS OR LOST DATA, COST OF PROCUREMENT
  OF SUBSTITUTE GOODS, TECHNOLOGY, SERVICES, OR ANY CLAIMS BY THIRD PARTIES
  (INCLUDING BUT NOT LIMITED TO ANY DEFENSE THEREOF), OR OTHER SIMILAR COSTS.

  Should you have any questions regarding your right to use this Software,
  contact Texas Instruments Incorporated at www.TI.com.
  */
```

craigm26, 2015-12-17:
I have no experience with them directly yet, but I would be a huge fan if we could do this. It's a very important development step for us (beaconsafety.co)

spoerk, 2016-11-26:
@craigm26 I issued a pull request of a first version of a BLE stack. https://github.com/contiki-os/contiki/pull/1936

Maybe this will help you.