contiki-os/contiki #1305: power consumption of TI sensortag 2650 demo

xianlin, 2015-10-13:
#### CC26xx-demo
Source Code is at https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-demo.c

This demo will read sensor reading and print values on the console, blink LED and advertise the BLE.

Supply sensortag with DC 2.7V (not coin cell battery), max current is 3mA and min current is 0.64mA.

When the GREEN LED ON, high current of 3mA, it last for 20 seconds ([as led toggle function starts to run](https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-demo.c#L384))

[image]

[image]![cc26xx-demo-high-current-02.jpg]

When the GREEN LED OFF, low current of 0.64mA (sometimes 0.8mA), now sensortag stops LED toggle, it also last for 20 seconds.

[image]![cc26xx-demo-low-current-01.jpg]

[image]

So if a coin cell battery of 220mAh, with duty cycle of 50% of 3mA and 0.7mA, it can last for about 5 days.

I will carry out next few tests soon. (e.g. 802.15.4 power consumption OR make the update/reading interval longer than 20 seconds, say 5 minutes). 

I will update the power measurement result here. Hopefully we can improve the power consumption and make the code running on the TI sensortag really **lean** and really low power.

cmorty, 2015-10-13:
You should also get hold of a scope. That might also give you some hints what to look for. Multimeters are only good for estimations. They are often inaccurate when the voltage/current changes a lot.

xianlin, 2015-10-13:
@cmorty I will try the scope next time. Thanks.

adamdunkels, 2015-10-13:
@jonnteolsson @g-oikonomou: there seems to be a lot of people seeing these problems, would be great if you guys could look into this given that the Sensortag platform is now one of the primary Contiki platforms.

bkozak-scanimetrics, 2015-10-13:
My PR #1257 should reduce the 0.64mA current significantly. On my CC26xx evaluation module I get a 40uA reduction in dee-sleep power consumption of the MCU.

g-oikonomou, 2015-10-13:
Bunch of things going on over the last few weeks, but this is in the ToDo!

jonnteolsson, 2015-10-17:
@adamdunkels we (@jonnteolsson and @g-oikonomou) will look into this issue starting Monday!

bkozak-scanimetrics, 2015-10-19:
@jonnteolsson and @g-oikonomou 

I've done some debugging of this myself and have found another problem which is causing high power consumption.

In lpm.c, you do not set an RTC wakeup if no etimer is running which means that we will wake on the next 128Hz clock interval. Ironically, we get higher power consumption if no etimer is running at all than if some etimer is scheduled.

xianlin, 2015-10-20:
I have conducted a **BLE firmware** power consumption experiment. 

The BLE firmware version is **1.20 (Jul 28 2015)** and the current is fluctuate between **0.5mA to 0.8mA** when idle/not connecting with master device and peak current with **3mA** when connecting with the master device.

The Firmware version:
[image]

Before connect, just powered on the BLE sensortag 2.0. (advertising interval 100ms and flash LED every 1 second)

[image]

Connecting by using mobile phone app via BLE scanning. (peak current last for about 2 or 3 seconds)

[image]

After BLE connection established and the andriod mobile app read the sensor value almost every 1000ms. (flutuate between 0.5mA and 0.8mA)

[image]

xianlin, 2015-10-20:
The below are the result of "**Very Sleepy Demo**" Power measurement by using a multi-meter.

https://github.com/contiki-os/contiki/tree/master/examples/cc26xx/very-sleepy-demo

When the `Mode=0`, sleepy mode is disabled, the current is high at around **2.84mA**

[image]

From COAP resource web page you can see the sleepy mode is disabled

[image]

Now enable the sleepy mode by issue a `POST` method to the COAP with message of `mode=1&interval=60&duration=20`

[image]

So the sensortag will sleep for 40s and wake up for another 20s, total 60s every time. 

The current consumption is at around **0.26mA**

[image]

So if we use duty ratio of **20s/40s every 1 minute**, the 240mAH CR2032 battery can last about **9 days**.

```
(0.26*40 + 2.8*20)/60 = 1.106667 mA
240/1.106667/24 = 9 days
```

if use **20s/100s every 120 seconds(2 minutes)**, it will last about 351 days which is almost **1 year**. (to be verified)

xianlin, 2015-10-20:
@bkozak-scanimetrics , your post at https://e2e.ti.com/support/wireless_connectivity/f/158/t/452029 mentioned there is a way to reduce the **deep sleep** current to about **1.5uA**.

Is your pull request https://github.com/contiki-os/contiki/pull/1257 being merged already? If it is merged, can it reduced the "Very Sleepy Demo" power consumption to 1.5uA when it is in deep sleep mode?

bkozak-scanimetrics, 2015-10-20:
@xianlin #1257 does implement the fix I mentioned on the TI forum. I don't know if anyone is considering merging it. Perhaps @g-oikonomou or @jonnteolsson has a better solution in mind for this problem.

I haven't been testing with the Very Sleepy Demo but I believe that #1257 must reduce the deep sleep consumption of the Very Sleepy Demo. As far as I know, there could be some other issue with that example which prevents sleeping at 1.5uA.

g-oikonomou, 2015-10-20:
Yes, #1257 is being considered. So is input in this thread here, other open related issues as well as various discussions in the mailing list. Thanks for the input, your patience is appreciated! :)

xianlin, 2015-10-21:
I have done the **cc26xx-web-demo** power measurement although using an oscilloscope is accurate but I used a multi-meter for simplicity.

Source Code

https://github.com/contiki-os/contiki/tree/master/examples/cc26xx/cc26xx-web-demo

Note

 - Disabled `net-uart` and `ble` function in the code

When LED is not blinking (`MQTT - Got TCP_DATA_SENT` in the debug output console), the current is below **0.47mA**. 

Here sensortag is **NOT** publishing mqtt messages.

[image]

When sensortag starts to publish MQTT message (`MQTT - Sending publish message! topic iot-2/evt/status/fmt/json topic_length 25` in the debug output console), the current is around **3mA** for 2 to 3 seconds) and LED blinks very shortly.

Then it goes back to around **0.47mA** and waits for another publish period.

[image]

If I turned off `wrapsix` NAT64 program running on the Boarder Router, the current shoot up to **8mA** while LED keeps on.

[image]

if I turned off `6lbr` program on Boarder Router, the current stays around **1.7mA** and LED keeps blinking.

[image]

### Conclusion

If `cc26xx-web-demo` is functioning properly (can publish MQTT message and display sensor reading on the IBM web dashboard) the current consumption is about **3mA** when publishing and **0.47mA** when NOT publishing.

To make this example/demo useful in the real world project, it should utilize **sleepy mode** to extend the battery life.

If the project is DC powered in some cases, it should avoid unnecessary radio messages OR reduce MQTT message length to get more bandwidth for other sensortags in the same network.

xianlin, 2015-10-21:
@bkozak-scanimetrics ,  I have done the `very-sleepy-demo` by using your PR #1257. 

The current reduced to **0.017mA = 17uA** compared to **0.26mA** with the previous experiment.

[image]

Thank you and it is impressive to get **1/15** deduction of the current.

Let's wait for the merge to happen and I will close this issue when it is merged.

g-oikonomou, 2016-01-09:
#1257 and a number of more recent pulls should be enough to close this one.

closed 2016-01-09 by g-oikonomou

vicgordienko, 2016-05-28:
@xianlin 

> So if we use duty ratio of 20s/40s every 1 minute, the 240mAH CR2032 battery can last about 9 days.
> 
> (0.26*40 + 2.8*20)/60 = 1.106667 mA
> 240/1.106667/24 = 9 days
> if use 20s/100s every 120 seconds(2 minutes), it will last about 351 days which is almost 1 year. (to be verified)

How did you calculate 351 days?
For my  calculation I get:
(0.26*100 + 2.8*20)/120 = 0.683333333 mA
240/240/0.683333333/24 = 9 days/24 = 14.63 days

xianlin, 2016-05-30:
@vicgordienko you are correct, it should be 14.63 days

Ashokrlz, 2017-05-27:
Hello everyone,

I'm searching for any documentation of Contiki talking about power consumption of wireless sensor nodes especially for CPU CC2650/CC1350? Regards,