contiki-os/contiki #1312: uip_split: Ignoring UIP_LLH_LEN?

cpboyd, 2015-10-19:
The current check for `uip_len >= UIP_SPLIT_SIZE + UIP_TCPIP_HLEN` seems to ignore `UIP_LLH_LEN` which is 0 in the default settings, but 14 for Ethernet.

Likewise, `tcplen = uip_len - UIP_TCPIP_HLEN;` again leaves out `UIP_LLH_LEN`.

vsaw, 2015-10-20:
Not sure if your observation is correct as `UIP_SPLIT_SIZE` is usually set to `UIP_TCP_MSS` to ensure TCP segment sizes are correct. See https://github.com/contiki-os/contiki/blob/master/core/net/ip/uip-split.c#L49

Therefore it must not take link layer headers into account as it completely operates on the TCP layer.

cpboyd, 2015-10-21:
I was aware that the default value of `UIP_SPLIT_SIZE` was `UIP_TCP_MSS` which has a default of `(UIP_BUFSIZE - UIP_LLH_LEN - UIP_TCPIP_HLEN)`.

However, my question is:  With a configurable split size via `UIP_SPLIT_CONF_SIZE`, does `UIP_LLH_LEN` need to be taken into account?

I guess, as you mentioned, since `uip_len` is the length of TCP data then it doesn't need to account for `UIP_LLH_LEN`.

laurentderu, 2015-10-21:
uip_len is not the length of TCP data, it's then length of the packet received, so if the MAC layer does not remove the link layer header, uip_len is equal to the size of the IP packet plus the link layer header.

So your custom UIP_SPLIT_CONF_SIZE must take into account UIP_LLH_LEN, also the split size calculated is indeed wrong if UIP_LLH_LEN != 0

I'm working on a PR fixing all the missing UIP_LLH_LEN (there are quite a few though)