contiki-os/contiki #1320: Rime examples won't work on the CC2650

wuxiao356, 2015-10-23:
I tried testing Rime protocol on CC2650 using the abc, broadcast, and unicast examples. But, none of them worked on the CC2650. In my case, everyone could send packet out, however, without any received. But both the sending and receiving function did work on CC2538. So is there any problem on the specific configuration on CC2650?

g-oikonomou, 2015-11-20:
We briefly tested Rime for CC26xx a while ago and it seemed to be working just fine.

If you think you have found a bug, please provide more information in order for me to try and reproduce it (version of sources used, example code, toolchain version to start with).

bkozak-scanimetrics, 2015-11-20:
I must have missed this issue when it first came out otherwise I would have pointed out that the problems with the ```LAST_FG_COMMAND_DONE``` IRQ which the radio driver had a couple months ago would have caused the symptoms that are described here.

g-oikonomou, 2016-05-14:
Tested successfully with fd15934, `example-broadcast` between srf06/cc26xx and sensortag/cc2650, both sending and receiving as expected.

closed 2016-05-14 by g-oikonomou

lemueljay, 2017-05-16:
Hi. Is the rime broadcast-example working in CC2538? I had a problem of receiving and sending of packets in CC2538. What might be the cause of this problem and how do I address this issue? Thank you.