contiki-os/contiki #1345: Contiki devices communicate to other wireless network devices running RIOT OS

xianlin, 2015-11-01:
Since contiki devices support 6lowpan and IP, is it possible to let Contiki devices talk with RIOT devices via IP network which is the same way your android smart phone talks to ios mobile devcies?

I understand this requires contiki and RIOT implement the same 6lowpan and IP protocol but I am NOT clear that if this is happening or happened?

There are 3 reasons the cross-OS communications over 6LowPan will fail:

1. Contiki follow strictly with 6LowPAN protocol but other OS don't. 
2. Other OS follow strictly with 6LowPAN protocol but Contiki don't. 
3. All OS follow strictly with 6LowPAN but the standards changes too frequently (not matured yet) and it causes differences when implementing the standards.

Just want to hear more thoughts about this cross-OS communication for IoT devices.

xianlin, 2015-11-03:
I only could found some relevant Q&A regarding the inter-operability.

http://stackoverflow.com/questions/12255291/are-6lowpan-stacks-in-tinyos-and-contiki-compatible

xianlin, 2015-11-03:
Looks like RIOT folks are trying to close the gap between Contiki

https://github.com/RIOT-OS/RIOT/issues/3662

OlegHahm, 2015-11-04:
see also discussion in https://github.com/RIOT-OS/RIOT/issues/4220

joakimeriksson, 2015-11-19:
We did some work on getting Contiki and Tiny OS to communicate with 6LoWPAN and RPL. That worked  quite ok! Not sure if we have the same ND as RIOT and some of the other things might also differ - but it seems like there have been some initial tests that indicates that it should work in the above RIOT-OS issues/discusssion. I do think it is a good thing to set up both OS:es for interop - we found lots of bugs while doing the Tiny OS and Contiki interops.

OlegHahm, 2015-11-19:
I would also appreciate a Contiki-RIOT interop very much. Maybe we can do so on the IoT-LAB testbed which should be supported by both platforms?

Yonezawa-T2, 2016-02-10:
When Contiki automatically configures its global IP address from Prefix Information option, it does not send DAO, so that router (RIOT) cannot build its routing table. I'm not sure this is standard compliant, but following patch works well between RIOT root-router and Contiki leaf.

https://github.com/Yonezawa-T2/contiki/commit/120aca099843492ab563863011911c71d27c959c

simonduq, 2016-02-10:
It should send a DAO when it first selects the node as preferred parent, doesn't it? Or are you talking about a runtime change of prefix? Do not hesitate to open a PR with your patch so we consider it for inclusion.

simonduq, 2016-02-10:
Btw, I should mention that we did test RIOT-Contiki interop at the last 6TiSCH plugest, see more discussion at: https://github.com/RIOT-OS/RIOT/issues/4220

OlegHahm, 2016-02-10:
Also, please be aware that I had to apply a patch similar to https://github.com/RIOT-OS/RIOT/pull/4732 on the RIOT site, because Contiki is not sending unicast DIS messages AFAIK.

Yonezawa-T2, 2016-02-10:
RIOT sends DIO with Prefix Information only if:

1. it sends DIO first time (gnrc_rpl_root_init), or 
2. it received unicast DIS (gnrc_rpl_recv_DIS)

So when Contiki is booted after RIOT is configured to be a root:

1. Contiki sends multicast DIS
2. RIOT sends DIO without Prefix Information
3. Contiki joins the given DODAG without global IP address
4. Contiki does not send DAO because no global address set for the node (dao_output)
5. Contiki sends unicast DIS when the probing timer is fired
6. RIOT sends DIO with Prefix Information
7. Contiki configures global IP address from the Prefix Information
8. Contiki does not send DAO because the parent nor its DTSN is changed (should_send_dao)

If Contiki is booted before RIOT is configured to be a root, RIOT sends DIO with Prefix Information and everything will be OK.

I'm not sure this is a bug of RIOT or Contiki. What is the correct behaviour the standards expects? How should we fix this?

Yonezawa-T2, 2016-02-10:
I didn't read #4732 before sending. That's it.

simonduq, 2016-02-10:
Contiki *does* send unicast DIS messages, by default every minute, until it joins any DAG.

simonduq, 2016-02-10:
Never worked with PIO-less DIOs, but it seems Contiki will join the DAG with no prefix, but then not send a unicast DIS as it probably should, indeed. Not exactly sure what the standards mandate here but sounds like a missing useful feature on the Contiki side anyway :)

g-oikonomou, 2016-02-10:
I echo Simon. Not sure what the RFCs say either, but perhaps we should ignore DTSN until we actually do have our own prefix? In the scenario above "new" (so to speak) DTSN would trigger DAO.

Yonezawa-T2, 2016-02-10:
> Contiki does send unicast DIS messages, by default every minute, until it joins any DAG.

It sends *multicast* DIS before joining DAG (handle_periodic_timer in rpl-timers.c).

After joining DAG, Contiki sends unicast DIS with random interval (handle_probing_timer in rpl-timers.c)

simonduq, 2016-02-10:
yes @Yonezawa-T2 I noticed my mistake straight after posting and edited. See next message ;)

Yonezawa-T2, 2016-02-10:
I should have reload browser before submitting. Sending unicast DIS for DIO without PIO seems nice.