contiki-os/contiki #1356: Contiki OS: no help / Comments / Documentation

kaleemullah360, 2015-11-04:
for beginner like me. it is very difficult to understand Contiki OS due to no help or documentation available.
i.e 
`SENSORS_ACTIVATE(battery_sensor);`
I don't know where this function came from what it require also from where does this `battery_sensor` come ?

or else suggest me the best source of help i.e tutorial / Docs

vsaw, 2015-11-04:
I get it that ContikiOS still has a steep learning curve but the code base is not to big to allow for effective searching. In your case: [GitHub Search Page 2 at the bottom](https://github.com/contiki-os/contiki/search?p=2&q=SENSORS_ACTIVATE%28x%29&type=Code&utf8=%E2%9C%93)

Apart from this the [Wiki](https://github.com/contiki-os/contiki/wiki) provides some help and the mailing list is also pretty active. If you still feel like things are not properly documented, please contribute to ContikiOS by fixing documentation.

alignan, 2015-11-04:
I've put together along with some friends a starter book heavily based on Contiki, as it is open source feel free to contribute as well:

https://github.com/alignan/IPv6-WSN-book/tree/rework

Chapter 3 talks about precisely about your query

kaleemullah360, 2015-11-04:
That really helpful, you're quick response is appreciated, I'll also contributes after some understanding. :+1:

closed 2015-11-04 by kaleemullah360

sieben, 2015-11-04:
@alignan Sounds like a great book. Should we post it on the Contiki wiki ?

alignan, 2015-11-04:
Sure, I'm cleaning some stuff towards the WALC 2015 workshop, I hope to get more stuff in, and expand the examples and applications.  I will be very happy for users to use and contribute.

sieben, 2015-11-04:
I've gone through a lot of slides provided by this conference and it's amazing :)

alignan, 2015-11-05:
[Done](https://github.com/contiki-os/contiki/wiki#getting-started), thanks for the suggestion!