contiki-os/contiki #1357: UDP send for unspecified listen ports?

cpboyd, 2015-11-04:
So, there doesn't seem to currently be a good way to send data back to an unknown port (e.g. DNS responses).

My modification to core/net/ipv4/uip.c (under `udp_send:`)
```c
  // If the remote port is 0 (i.e. any port), set destport to the port we just received from.
  if (uip_udp_conn->rport == 0) {
	  BUF->destport = BUF->srcport;
  } else {
	  BUF->destport = uip_udp_conn->rport;
  }

  // If the remote IP is the zero address (i.e. any IP), set the destipaddr to the IP we just received from.
  if (uip_ipaddr_cmp(&uip_udp_conn->ripaddr, &uip_all_zeroes_addr)) {
	  uip_ipaddr_copy(&BUF->destipaddr, &BUF->srcipaddr);
  } else {
	  uip_ipaddr_copy(&BUF->destipaddr, &uip_udp_conn->ripaddr);
  }

  BUF->srcport  = uip_udp_conn->lport;

  uip_ipaddr_copy(&BUF->srcipaddr, &uip_hostaddr);

  uip_appdata = &uip_buf[UIP_LLH_LEN + UIP_IPTCPH_LEN];
```

vsaw, 2015-11-04:
It I understand correctly [`uip_udp_packet_sendto`](https://github.com/contiki-os/contiki/blob/master/core/net/ip/uip-udp-packet.h) does exactly what you want.

cpboyd, 2015-11-04:
@vsaw Not quite.  You still have to specify the IP address and port to respond to.

I've instantiated a UDP port with IP address of NULL and remote port of 0. (And attached it to the local port 53.)

My specific use case is handling DNS queries and issuing responses.
I'm handling the UDP request (via `UIP_UDP_APPCALL()`) and performing processing on the data buffer to issue a response.

However, there's no clear way to determine WHAT IP address or port number that the request came from.  Thus, if I try a simple `uip_udp_send(dataSize);` it'll try to send the UDP packet to 0.0.0.0:0

vsaw, 2015-11-05:
I was actually suggesting you use `uip_udp_packet_sendto` there you can specifically give it the target address and port. See https://github.com/contiki-os/contiki/blob/master/core/net/ip/uip-udp-packet.h#L46#L47

If your question is how to get the source IP and Port, you can read them directly from `uip_buf`. This simple line should work:

```C
#define UIP_IP_BUF ((struct uip_ip_hdr *)&uip_buf[UIP_LLH_LEN])
#define UIP_UDP_BUF ((struct uip_udp_hdr *)&uip_buf[UIP_LLH_LEN + UIP_IPH_LEN])

PRINT6ADDR(&UIP_IP_BUF->srcipaddr);
PRINTF("%D", HTONS(UIP_UDP_BUF->srcport));
```

schuschu, 2015-11-05:
As this seems a good time to mention this, the ipv6-udp example (upd-server) listens to a port that is not 0. Not sure if this is intended but as far as I can tell it does not work that way (compare with the upd example int the cc2538 folder)

cpboyd, 2015-11-11:
@vsaw Ah, I see. I'm aware that I could use that as a work-around, but is there a specific reason that we can't more easily send a response to a UDP connection?

vsaw, 2015-11-11:
@cpboyd My guess is that the UDP API of Contiki is still pretty much a legacy of the "old Wireless Sensor Network days" where it was enough to get data out of a device. The requirement for device to act as a receiving server where not there yet. But it's just a guess.

In any case I hope this solves your problem.

g-oikonomou, 2015-11-11:
With the newer UDP API (`simple-udp.h`) you define a receive callback. When it gets called, the core will also give you the remote UDP port as one of the callback's arguments. You can then simply send to this port. Examples of this approach under `examples/ipv6/simple-udp-rpl/unicast-receiver.c`

With the older UDP API, you retrieve the remote UDP port from the uIP buffer. You set the UDP "connection's" remote port to this value, you send your reply, you set `rport` back to 0 in order to be able to receive the next message. One example of this usage in `examples/cc2538dk/udp-ipv6-echo-server/udp-echo-server.c`. Something like:

```
    uip_ipaddr_copy(&server_conn->ripaddr, &UIP_IP_BUF->srcipaddr);
    server_conn->rport = UIP_UDP_BUF->srcport;
    uip_udp_packet_send(server_conn, buf, len);
    uip_create_unspecified(&server_conn->ripaddr);
    server_conn->rport = 0;
```

cpboyd, 2015-11-13:
Ok.  If this is intentional behavior for udp_send on open UDP listen ports, then I'll just close this.

There's plenty of workarounds, but this behavior just seemed counter-intuitive to me.

closed 2015-11-13 by cpboyd