contiki-os/contiki #1367: should off_t type be signed or unsigned? In MSP430 Z1 Platform

myrfy001, 2015-11-13:
When I using MSP430-GCC, which is the new complier published by TI as the successor of the community driven MSPGCC, I meet the question bellow:

```C
#include "contiki.h"
#include <stdio.h>
```

in contiki.h, it includes the platform-conf.h and it deffines off_t as unsigned long.
in stdio.h, it includes types.h given by Texas Instruments and it defines off_t as long.

Besides Texas Instruments, Linux also defines off_t as long. __So, should the off_t be changed to signed long instead of unsigned long?__

cmorty, 2015-11-13:
If I understand things right, according to POSIX it should be signed: http://pubs.opengroup.org/onlinepubs/7908799/xsh/systypes.h.html

myrfy001, 2015-11-14:
So, the problems comes...
When compiling contiki with some other code using POSIX, there will be a type conflict.
Will change contiki's off_t to signed have any bad effects?

cmorty, 2015-11-15:
My guts tell me it shouldn't have a negative effect. So I suggest to change it, make a pull-request, and see what happens. After all the regression testing is quite comprehensive. Also PRs normally get more attention than issues.

g-oikonomou, 2015-11-20:
My 2c: if a type is defined in toolchain headers, we shouldn't overwrite said define unless there is a really really really good reason to do so. Even more so if standards have an explicit opinion.

alignan, 2015-11-21:
@g-oikonomou agree.  Does this break anything? can you `grep` any potential bug?

alignan, 2016-03-25:
Closing, if you want to create a PR to address the change we can review later.

closed 2016-03-25 by alignan

simonduq, 2016-03-25:
I agree this type is supposed to signed, and btw the PR is already merged :) https://github.com/contiki-os/contiki/pull/1375

alignan, 2016-03-25:
Oh nice!