contiki-os/contiki #1373: [CCxxxx] TI DSLite and easy JTAG uploading for Sensortag (and others) in Contiki

JelmerT, 2015-11-16:
After playing around with TI's [online dev environment](https://dev.ti.com/ide) and a tip from a Contiki-user, I stumbled on the `DSLite` command line program.

DSLite is installed locally by the `TICloudAgent` (which is the interface between the browser plugin and the local hardware). DSLite seems to be able to program targets via the XDS-110 and XDS-100 emulators, when you give it the correct config file and drivers (which are defined in the config files of the chip you're programming). It is fully controllable through websockets (that's how it's controlled by the cloud9 dev environment), and it also seems to support basic config via cmd line arguments (`load` and `runWithCIO`). I tested it on OSX but I'm presuming there's a windows / linux version too.

So this means there's usable cross platform drivers for the XDS JTAG emulators -> hurray!

Now is the question: would it make sense to write a wrapper around DSLite for easy, one command, uploading / debugging in Contiki? Is there a need for this? How would this be integrated in the source code (outside tool or copy in the repo?)? And how would this work copyright wise? Maybe someone from TI (@jonnteolsson ?) could shed some light on the tool itself, and TI's plans on releasing the docs or code of the tool?

TL;DR: Is a wrapper around DSLite for Contiki useful / possible / wanted?

philvalencia, 2015-11-16:
I've used the DSLite from the command line on my mac to flash the SensorTag (CC2650) using the http://www.ti.com/tool/cc-devpack-debug (USD$15) programmer (XDS110). Haven't yet checked on how easy it is to generate the compatible .out format from Contiki - I assume it would be either already supported or fairly trivial to add to the make script to allow this method as a programming method. Note that I also needed the CC2650F128.ccxml file which TICloudAgent downloaded the first time I used that. Are the contiki crew working on OTA reprogramming for SensorTags? I had the feeling they were, but not sure where I got that idea from.

schuschu, 2016-01-08:
With the latest update of energia there is now support for cc2650 in the "official" dslite. I have yet to try it out but it seems promising (as there is a release for windows/linux/mac). Has anyone tried this so far?

JelmerT, 2016-03-25:
The latest version of Energia seems interesting indeed. But it doesn't seem that there is any interest for this feature at this time.

I'm closing this for now, feel free to comment to re-open.

closed 2016-03-25 by JelmerT