contiki-os/contiki #1384: payload format not fixed

bbhagasra1, 2015-11-20:
fix payload data format ...........it is HEX or ASCII ??

msloth, 2015-11-20:
It's neither and both. It's just data, binary data. Hex and ASCII are just ways to represent the data. Ie, ASCII 'A' is hex 0x41 is decimal 65 is binary 1000001.

Google eg `data representation`.

closed 2015-11-20 by simonduq

bbhagasra1, 2015-11-21:
Hi,
       If input to node(serial port UART) is HEX and node sending over 
the air ASCII then need to covert hex to ascii because receiver node 
understand ASCII at RF channel and again change to ASCII to HEX for 
interface Device(HEX CMD input).

for example :- 0x00 in HEX have value 0 if at RF node(ASCII data format 
used) then value is NULL.

so we need to define input (UART) and RF channel data format same.