contiki-os/contiki #1386: Wake on radio mac/radio layers

tussock-cam, 2015-11-20:
Has there been any discussion about building mac and radio layers around Enhanced Wake on Radio (eWOR) that's available in recent ccxxxx chips? My company's looking to reduce the power consumption we're getting with ContikiMAC + cc1200, and it seems like wake or radio would be a good way to go. We've used to successfully in previous projects, albeit without full net stacks or complex protocols.

So, is there any reason why wake on radio wouldn't be successful in Contiki? We'd like to get a bit of feedback before investing any serious time into it!

Thanks

g-oikonomou, 2015-11-20:
I cannot immediately think of any reasons why eWOR would not work with Contiki. If I were to implement it, I'd start with NullRDC and then I'd spend some time deciding whether other MAC/RDC layers are suitable or whether something new would have to be devised.

bkozak-scanimetrics, 2015-11-20:
Hello @tussock-cam, I've recently created a fresh implementation of ContikiMAC which allows more responsibility to be placed on the underlying radio hardware. Although I don't have any support for WOR or CC1200 at this time, I think this may be a better starting point than the existing ContikiMAC driver. In my opinion, the existing ContikiMAC architecture is not very good at dealing with special hardware features such as this one.

If you are interested in reducing power consumption, you might also be interested to know that my new implementation has reduced power consumption on the platform I'm currently spending most of my time on (CC2650).

Since we haven't released anything publicly yet, send me an email if you're interested in knowing more (it's public on my github home page). Maybe we could work together on this.

tussock-cam, 2015-11-22:
@bkozak-scanimetrics your ContikiMAC implementation sounds interesting, what hardware features have you been able to take advantage of? I'm not sure it would be a great starting spot for eWOR, though, as I'd imagine the radio driver doing most of the work and just using nullrdc.

bkozak-scanimetrics, 2015-11-23:
> as I'd imagine the radio driver doing most of the work and just using nullrdc

I had imagined eWOR being used as a stand-in for ContikiMAC's periodic CCA checks.  Perhaps I am misunderstanding what eWOR is capable of but I had thought that it could work quite well for this. 

I think my implementation of ContikiMAC would be a better starting off point for this mainly because it handles asynchronous multi-tasking better, using actual locks, atomic sections, and callbacks so that the hardware can work in the background. For example, in my driver ContikiMAC inserts a callback into the receive interrupt of the underlying radio driver so that, during the periodic CCA check, it can turn the radio back off immediately after getting a packet; there is a lot of synchronization that needs to happen in order to do this without causing a race condition. By contrast, the current driver will only turn the radio back off after enough time has passed.

If you think that you just need a nullrdc implementation, however, then my code won't be of any help to you.

> what hardware features have you been able to take advantage of?

So far we haven't got anything really special working; just the basic transmit, receive, detect ACK, and CCA functionality. Unfortunately, we've found that some of the really advanced features of the CC2650 MCU which we wanted to add support for just don't work well with ContikiMAC.

jesse-tussock, 2015-11-23:
Hmm, I hadn't really considered how I'd go about doing CCA checks, but eWOR would be useful for more than just that. We wouldn't need to enter rx mode at fixed intervals, we'd just set the radio into eWOR mode and wait for packets to arrive. When we want to transmit, do cca, enter tx and when done go back into eWOR. Most of ContikiMAC's core wouldn't be needed. 

Your ContikiMAC modifications definitely sound interesting, though. Do you have an estimates for how much you improve efficiency by?

bkozak-scanimetrics, 2015-11-23:
Although we have hard numbers for our particular platform and applications, It's difficult to give a general efficiency improvement estimate due to the nature of the changes. Most of the improvement comes from:

1. The CPU is allowed to perform other tasks or sleep while the radio is transmitting and/or receiving
2. Phase optimization is now working really well (provided that we transmit sufficiently often so as to keep the nodes in sync)

As a result, the efficiency improvement is going to depend heavily on what exactly it is that your application does while TX and RX are on-going, the time it takes to transmit, how quickly your MCU can go back to sleep after waking etc.

closed 2016-04-06 by g-oikonomou

saurabhbansal90, 2017-05-16:
@bkozak-scanimetrics I am also working on same hardware CC2650. Can you please provide your changes in ContikiMAC for making use of eWOR.
I tried to mail you, but your email id bkozak@scanimetrics.com seems not to be working.