contiki-os/contiki #1401: uIPv6 - uIP needs to be aligned w.r.t LLHLEN and not start of uIP

joakimeriksson, 2015-11-27:
Recently there have been a few fixes that have been doing a bit strange alignment fixes for data in IPv6 packets that are supposed to be aligned but seems not to be. This is likely due to our efforts to align uip_buf is failing in some cases (LLHLEN having an odd number would likely cause this). I expect crashes to come from anywhere in the stack if we do not keep IPv6 packets aligned since we cast to data to structs all the time.

Im am referring to PR: #1207 to be specific (and #1394).

I do believe that there is a bug somewhere else - either bad options generated with wrong size or the whole uipbuf start is not aligned due to LLHLEN being set to a value that cause IPv6 packets to come in unaligned.

simonduq, 2015-11-27:
I don't think these are strange fixes. I believe `uip_buf` to be aligned at all times. What is not, in the two PR above, is `nd6_opt_llao[UIP_ND6_OPT_DATA_OFFSET]`, not `uip_buf` per se. And that's ok, I don't think that particular field is supposed to be aligned.

I don't see the issue, am I missing something?

simonduq, 2015-11-27:
Or are you talking about problems in the Eth case? Else, `UIP_LLH_LEN` is always 0 anyway, right?

joakimeriksson, 2015-11-27:
For 6LoWPAN UIP_LLH_LEN is 0 - in other cases (when it is not) we will risk ending up with non-aligned uip_buf which needs to be fixed. And the nd6_opt_llao should be aligned at 2-byte boundary - option is aligned at something (8-byte) and it starts 2-bytes in. But if there is need for 32-bit alignment for lladdress it might be an issue? Was there a 32-bit requirement on the copy or something else?

laurentderu, 2015-11-27:
On CC2538 (an I believe on all Cortex-M3), you must have 32 bits alignment for struct copy. with UIP_LLH_LEN == 14 the uIP stack crashes, usually in nd6. I tried to remove the bad copy, but due to lack of time, I eventually set UIP_LLH_LEN to 16 as temporary workaround.
Also, some parts of the code do not use UIP_LLH_LEN, I fixed that part, but also I did not have time to make a proper PR.

simonduq, 2015-11-27:
@joakimeriksson yes on some platforms you need pointers to be word-aligned, i.e. 32-bit aligned on 32-bit platforms. jn516x is an example but like @laurentderu, I also think Cortex-M3 has the same requirement. And fair enough, if we still want to support Ethernet then a fix is needed I agree. Maybe simply as you suggest, by aligning `uip_buf + UIP_LLH_LEN` rather than `uip_buf`.

joakimeriksson, 2015-11-27:
@simonduq - ok yes that is a problem then with only this specific case (if we assume uip_buf + UIP_LLH_LEN is aligned). We should probably move the alignment extraction to the linkaddr.c (and when are at it just ensure that we have one type of lladdr) - now it is both risky and confusing with both uip_lladdr and linkaddr.

simonduq, 2015-11-27:
:+1: on having a single lladdr type!
For extraction I don't know, this really a simple helper function, with most of the code LLAO-specific.

joakimeriksson, 2015-11-27:
The thing is that any place where you construct a linkaddr from byte-array (e.g. rime, or anything else) should use the same function. It would be good to have including a clear comment so that anyone that create / copy know what they are doing - even if using something that is not ND6.

simonduq, 2015-11-27:
Right but that holds for any struct, so I don't think having a linkaddr-specific extractor helps a lot. I also don't think there are (many) such bugs left in the code-base; have you spotted any other one? For what it's worth, we get a runtime exception on jn516x for every unaligned access, and this hasn't happened for a very long time now.