contiki-os/contiki #1415: mqtt-demo.c:56:32: fatal error: dev/cc2538-sensors.h: No such file or directory

kaleemullah360, 2015-12-10:
In Cooja Simulator:
1. Create new simulation
2. add Z1 mote
3. brows to `/home/kaleem/contiki/examples/cc2538dk/mqtt-demo/mqtt-demo.c`, select and compile the file.
**Full Stack Trace:**

> make mqtt-demo.z1 TARGET=z1 
mkdir obj_z1
  CC        ../../../apps/mqtt/mqtt.c
  CC        ../../../platform/z1/./contiki-z1-platform.c
  CC        ../../../cpu/msp430/f2xxx/msp430.c
  CC        ../../../core/dev/leds.c
  CC        ../../../cpu/msp430/./watchdog.c
  CC        ../../../platform/z1/dev/xmem.c
  CC        ../../../cpu/msp430/f2xxx/spi.c
  CC        ../../../dev/cc2420/cc2420.c
../../../dev/cc2420/cc2420.c:528:3: warning: initialization from incompatible pointer type [enabled by default]
../../../dev/cc2420/cc2420.c:528:3: warning: (near initialization for ‘cc2420_aes_128_driver.set_key’) [enabled by default]
../../../dev/cc2420/cc2420.c: In function ‘cc2420_transmit’:
../../../dev/cc2420/cc2420.c:659:24: warning: variable ‘sfd_timestamp’ set but not used [-Wunused-but-set-variable]
  CC        ../../../platform/z1/dev/cc2420-arch.c
  CC        ../../../cpu/msp430/./cc2420-arch-sfd.c
../../../cpu/msp430/./cc2420-arch-sfd.c: In function ‘cc2420_timerb1_interrupt’:
../../../cpu/msp430/./cc2420-arch-sfd.c:44:7: warning: variable ‘tbiv’ set but not used [-Wunused-but-set-variable]
  CC        ../../../platform/z1/./node-id.c
  CC        ../../../core/lib/sensors.c
  CC        ../../../platform/z1/dev/button-sensor.c
  CC        ../../../core/cfs/cfs-coffee.c
  CC        ../../../platform/z1/dev/radio-sensor.c
  CC        ../../../cpu/msp430/f2xxx/uart0.c
  CC        ../../../cpu/msp430/dev/uart0-putchar.c
  CC        ../../../cpu/msp430/./uip-ipchksum.c
  CC        ../../../core/dev/slip.c
  CC        ../../../cpu/msp430/./slip_uart0.c
  CC        ../../../platform/z1/dev/z1-phidgets.c
  CC        ../../../dev/sht11/sht11.c
../../../dev/sht11/sht11.c: In function ‘sht11_init’:
../../../dev/sht11/sht11.c:218:4: warning: #warning SHT11: DISABLING I2C BUS [-Wcpp]
  CC        ../../../dev/sht11/sht11-sensor.c
  CC        ../../../platform/z1/dev/light-sensor.c
  CC        ../../../platform/z1/dev/battery-sensor.c
  CC        ../../../platform/z1/dev/sky-sensors.c
  CC        ../../../platform/z1/dev/tmp102.c
  CC        ../../../platform/z1/dev/temperature-sensor.c
  CC        ../../../platform/z1/dev/light-ziglet.c
  CC        ../../../platform/z1/dev/relay-phidget.c
  CC        ../../../platform/z1/dev/tlc59116.c
  CC        ../../../platform/z1/dev/sht25.c
  CC        ../../../core/lib/me.c
  CC        ../../../core/lib/me_tabs.c
  CC        ../../../core/lib/crc16.c
  CC        ../../../platform/z1/dev/i2cmaster.c
  CC        ../../../platform/z1/dev/adxl345.c
  CC        ../../../cpu/msp430/./flash.c
  CC        ../../../cpu/msp430/f1xxx/clock.c
  CC        ../../../cpu/msp430/./leds-arch.c
  CC        ../../../cpu/msp430/./lpm.c
  CC        ../../../cpu/msp430/./mtarch.c
  CC        ../../../cpu/msp430/f1xxx/rtimer-arch.c
  CC        ../../../core/loader/elfloader.c
  CC        ../../../core/loader/elfloader-msp430.c
  CC        ../../../core/loader/symtab.c
  CC        ../../../core/sys/arg.c
  CC        ../../../core/sys/autostart.c
  CC        ../../../core/sys/compower.c
  CC        ../../../core/sys/ctimer.c
  CC        ../../../core/sys/energest.c
  CC        ../../../core/sys/etimer.c
  CC        ../../../core/sys/mt.c
  CC        ../../../core/sys/process.c
  CC        ../../../core/sys/procinit.c
  CC        ../../../core/sys/rtimer.c
  CC        ../../../core/sys/stimer.c
  CC        ../../../core/sys/timer.c
  CC        ../../../core/dev/nullradio.c
  CC        ../../../core/dev/serial-line.c
  CC        ../../../core/lib/aes-128.c
  CC        ../../../core/lib/assert.c
  CC        ../../../core/lib/ccm-star.c
  CC        ../../../core/lib/gcr.c
  CC        ../../../core/lib/ifft.c
  CC        ../../../core/lib/list.c
  CC        ../../../core/lib/memb.c
  CC        ../../../core/lib/mmem.c
  CC        ../../../core/lib/petsciiconv.c
  CC        ../../../core/lib/print-stats.c
  CC        ../../../core/lib/random.c
  CC        ../../../core/lib/ringbuf.c
  CC        ../../../core/lib/settings.c
  CC        ../../../core/lib/trickle-timer.c
  CC        ../../../core/net/ipv6/sicslowpan.c
  CC        ../../../core/net/ipv6/uip6.c
  CC        ../../../core/net/ipv6/uip-ds6.c
  CC        ../../../core/net/ipv6/uip-ds6-nbr.c
  CC        ../../../core/net/ipv6/uip-ds6-route.c
  CC        ../../../core/net/ipv6/uip-icmp6.c
  CC        ../../../core/net/ipv6/uip-nd6.c
  CC        ../../../core/net/ip/dhcpc.c
  CC        ../../../core/net/ip/ip64-addr.c
  CC        ../../../core/net/ip/psock.c
  CC        ../../../core/net/ip/resolv.c
  CC        ../../../core/net/ip/simple-udp.c
  CC        ../../../core/net/ip/slipdev.c
  CC        ../../../core/net/ip/tcpip.c
  CC        ../../../core/net/ip/tcp-socket.c
  CC        ../../../core/net/ip/udp-socket.c
  CC        ../../../core/net/ip/uip-debug.c
  CC        ../../../core/net/ip/uiplib.c
  CC        ../../../core/net/ip/uip-nameserver.c
  CC        ../../../core/net/ip/uip-packetqueue.c
  CC        ../../../core/net/ip/uip-split.c
  CC        ../../../core/net/ip/uip-udp-packet.c
  CC        ../../../core/net/rpl/rpl.c
  CC        ../../../core/net/rpl/rpl-dag.c
  CC        ../../../core/net/rpl/rpl-dag-root.c
  CC        ../../../core/net/rpl/rpl-ext-header.c
  CC        ../../../core/net/rpl/rpl-icmp6.c
  CC        ../../../core/net/rpl/rpl-mrhof.c
  CC        ../../../core/net/rpl/rpl-of0.c
  CC        ../../../core/net/rpl/rpl-timers.c
  CC        ../../../core/net/linkaddr.c
  CC        ../../../core/net/nbr-table.c
  CC        ../../../core/net/netstack.c
  CC        ../../../core/net/packetbuf.c
  CC        ../../../core/net/queuebuf.c
  CC        ../../../core/net/mac/csma.c
  CC        ../../../core/net/mac/frame802154.c
  CC        ../../../core/net/mac/framer-802154.c
  CC        ../../../core/net/mac/framer-nullmac.c
  CC        ../../../core/net/mac/mac.c
  CC        ../../../core/net/mac/mac-sequence.c
  CC        ../../../core/net/mac/nullmac.c
  CC        ../../../core/net/mac/nullrdc.c
  CC        ../../../core/net/mac/nullrdc-noframer.c
  CC        ../../../core/net/mac/phase.c
  CC        ../../../core/net/mac/contikimac/contikimac.c
  CC        ../../../core/net/mac/contikimac/contikimac-framer.c
  CC        ../../../core/net/llsec/anti-replay.c
  CC        ../../../core/net/llsec/ccm-star-packetbuf.c
  CC        ../../../core/net/llsec/nullsec.c
cp ../../../tools/empty-symbols.c symbols.c
cp ../../../tools/empty-symbols.h symbols.h
  CC        symbols.c
  AR        contiki-z1.a
  CC        mqtt-demo.c
mqtt-demo.c:56:32: fatal error: dev/cc2538-sensors.h: No such file or directory
compilation terminated.
make: *** [mqtt-demo.co] Error 1
Process returned error code 2

Help me to solve the problem please.

msloth, 2015-12-10:
You are trying to run a cc2538dk example as another platform. Either comment out the cc2538dk stuff until it works, or don't run it at all. The cc2538dk sensors are not possible to simulate in Cooja.

kaleemullah360, 2015-12-10:
as I am beginner, doing project in Porting MQTT in Contiki Cooja simulator. so please guide me, help me. give me some links papers any thing so that I can also participate in Contiki OS development.
I'll be very thankful to you. :+1:

tonnenpinguin, 2015-12-10:
What @msloth was trying to tell you is that you are using code that is intended to be used with cc2538 with another platform (z1), which has not implemented the cc2538 specific sensors. This is also why you are getting this error.
I suggest you start with the contiki [wiki](https://github.com/contiki-os/contiki/wiki) as well as the great [IOT in 5 days](https://github.com/alignan/IPv6-WSN-book) book.

kaleemullah360, 2015-12-10:
@tonnenpinguin I understand now, so many thanks to both @msloth and @tonnenpinguin  for helping me. can you direct me to some resource or guy who is developing systems in MQTT, CoAP in Contiki  OS and Cooja Simulator. I can't afford hardware.

closed 2015-12-10 by kaleemullah360

saeedsobhaninejad, 2018-11-22:
hi
can you solve this matter?
i have this problem in cooja
and i can't solve it.
please help me.
Thank you