contiki-os/contiki #1421: enc28j60: Possible Ethernet -> RF overrun

bthebaudeau, 2015-12-14:
`enc28j60_ip64_driver_process()` unconditionally passes all the Ethernet packets to `IP64_INPUT()` (`ip64_eth_interface_input()`) almost as soon as they are received. Ethernet Rx being way faster than RF Tx, this can result in a temporary RF buffer overrun in the IP64 device. When this happens for IP packets, `ip64_eth_interface_input()` makes `ip64_4to6()` overwrite `uip_buf[]`, possibly before the actions resulting from the previous call to `tcpip_input()` are finished with the previous contents of this buffer. This then results in "sicslowpan input: Dropping 6lowpan packet that is not a fragment of the packet currently being reassembled" errors on mesh nodes.

I typically got these errors when performing OTA firmware upgrade on CC2538 with an application based on Mist. I no longer have this hardware platform, but the software modules that were causing this issue in Mist seem to be almost the same in the current Contiki master, so this issue should still be valid.

I had suggested a workaround for this in https://github.com/contiki-os/contiki/commit/30b3041e72b48b314442685adbf1a09a08b8bdfc (initially in #1168), but it is neither a solution nor universal, so I open this issue instead.

adamdunkels, 2015-12-14:
I don't think this was caused by the Ethernet driver per se, but rather because of some old bug in either the `csma.c` code (where there have been a few since back then) or in the `sicslowpan.c` code (in which there have been multiple bugs like this fixed too). We're doing a lot of firmware updates using the ENC28j60 driver and onto the radio and haven't had any issues like these in the current/recent codebase.

The `uip_buf` buffer is only accessed in a single-threaded fashion, so new data should never overwrite anything there. The only way this could happen would be if code called from an interrupt handler was altering the `uip_buf`, which would be a serious problem, but the ENC28j60 driver does not use interrupts.

bthebaudeau, 2015-12-14:
OK, so I close this issue. It's here for the record anyway if ever needed.

closed 2015-12-14 by bthebaudeau