contiki-os/contiki #1431: cc26xx-web-demo on sensortag change MQTT configuration failed with "Content-Length Required" message

xianlin, 2015-12-21:
When compile and run the latest cc26xx-web-demo, I want to change the mqtt broker from IBM quickstart to IBM registered device, in the [ MQTT/IBM Cloud Config ] tab I entered valid strings for those fields and when I click submit button, the web page shows error message "**Content-Length Required**".

Does anyone encountered the same issues? I remember a few months back I still can change the mqtt configuration through this web interface but now with new commit it failed.

Anyone has been successfully tested connect and publish to **IBM registered device MQTT broker**?
Thanks.

jonnteolsson, 2015-12-22:
Did you try using different browsers?

jonnteolsson, 2015-12-22:
and yes, I have been using registered devices on IBM IoT Foundation cloud service.

g-oikonomou, 2015-12-22:
The web server requires a Content-length header, otherwise the logic to determine the end of an incoming POST message becomes a fair bit more complicated. If Content-length is not there, you get this error.

Either the browser is not sending Content-length, or it is but the tag somehow fails to parse the header correctly. It should be easy to figure out which of the two by using the browser's debugging features which will basically show you exactly what headers are being sent in the POST message.

svnesbo, 2016-01-13:
I encountered the same problem at some point on a cc2650 sensortag.
I had around 10 different sensortags that I were using in a mesh network, and oddly enough this happened on only one of them. The other sensortags were fine. Reflashing it did not help either.
I also tried different both firefox and internet explorer, and I got the same error using both browsers.

Also, I could be wrong, but I seem to remember that the sensortag did update the fields when I submitted the form although I got the Content-Length Required error after the form had been submitted.

Unfortunately I don't have any more information at this point, but if I encounter this again I will make some notes here.

g-oikonomou, 2016-01-13:
This may come across as totally random, but please try adding a 0 somewhere in the new broker address text box. So if for example you are getting the error while trying to set broker address to `aaaa::345:6789:a:1`, try setting `aaaa::345:6789:a:01`

alignan, 2016-03-25:
@svnesbo: did @g-oikonomou suggestion worked for you?

svnesbo, 2016-03-26:
Sorry I forgot to update this thread. I did see this error again, but I wasn't able to make it go away by adding leading zeros. I haven't used the web demo application for a while and I haven't had time to take a deeper look at it.

g-oikonomou, 2016-05-14:
I just re-tested the MQTT configuration page on a cc2650 sensortag and it worked fine for me. Using Firefox on OS X. See #1594 for more info on my setup.

g-oikonomou, 2016-05-14:
I am closing this for now, since I cannot reproduce. We can re-open if more info becomes available.

closed 2016-05-14 by g-oikonomou