contiki-os/contiki #1437: jn516x: vMMAC_SetChannelAndPower

neophob, 2015-12-28:
I use a jn5168 module with an external FEM to boost the tx signal. I compile contiki using the parameter ```-DRADIO_TEST_MODE=RADIO_TEST_MODE_HIGH_PWR```.

I suspect the sending power of my node is too low, as the communication starts to fail after about 7m (using only one node). micromac-radio.c sets the tx power by calling the jn516x API call ```vMMAC_SetChannelAndPower(current_channel, current_tx_power);``` - however I cant find this reference in my SDK - I'm using Build Number 1281.

Which SDK do you use and where can I find this API call?

TeVeDe, 2016-01-11:
The function is not in MMAC.h as one would expect, but is defined 'ext'. I use the SDK with build number 1307.

neophob, 2016-01-11:
thanks, I couln't find a reference in the JN SDK docs, but the function is (obviously) available

closed 2016-01-11 by neophob