contiki-os/contiki #1455: many performance bugs and security bugs in contiki2.7 and contiki3.0

zq301, 2016-01-11:
I have test contiki2.7 and contiki3.0 source code, and I found some performance bugs and security bugs in contiki.
1.there are many while or for loop structures that represent the same function in contiki source code, these repetition can be removed by some ways. In addition, some of these structures can be improved to save computing time, both functions parse the string byte wise and are quite slow for longer strings  may be improved by add the byte wise to 16 bits wise or more In an appropriate way.
    According to the function of loop structures, I classified the loop structures I recently focus on to   five classes as follows:
	Shave off any leading spaces from the beginning of a string .(classification 1)
	Shave off spaces from the end of a string. (classification 2)
	Find first space in a string, then replace the space with zero. (classification 3)
	Shave off zero from the beginning of a string. (classification 4)
	If a character in a string is not "0xff", then return(In the following loops of the function are all in "is_broadcast_addr" process). (classification 5)

In contiki2.7 and contiki3.0,all the loops of the five classes are in the table in the following. The table also deccrible the location and classification of these loops.
number	Bug location	Line number	classification
1	Contiki2.7/examples/ipv6/rpl-collect/collect-common.c	129	1
2	Contiki2.7/examples/ipv6/rpl-collect/collect-common.c	137	1
3	Contiki2.7/apps/shell/shell.c	247	1
4	Contiki2.7/apps/shell/shell.c	531	1
5	Contiki2.7/apps/shell/shell-download.c 	150	1
6	Contiki2.7/apps/shell/shell-sendtest.c	138	1
7	Contiki2.7/apps/shell/shell-sendtest.c	148	1
8	Contiki2.7/apps/shell/shell-time.c	252	1
9	Contiki2.7/apps/shell/shell-time.c	327	1
10	Contiki2.7/apps/shell/shell-netperf.c	576	1
11	Contiki2.7/apps/irc/ircc.c	216	1
12	Contiki2.7/apps/webbrowser/www.c	300	2
13	Contiki2.7/apps/email/email.c	188	3
14	Contiki2.7/cpu/6502/ipconfig/ipconfig.c	132	3
15	Contiki2.7/apps/netconf/netconf.c	128	3
16	Contiki2.7/apps/telnet/simpletelnet.c	160	3
17	Contiki2.7/tools/z80/hex2bin/src/hexameter.c	95	4
18	Contiki2.7/examples/ipv6/slip-radio/no-framer.c	70	5
19	Contiki2.7/core/net/mac/framer-802154.c	82	5
20	Contiki2.7/core/net/mac/sicslowmac.c	89	5
21	Contiki3.0/examples/ipv6/rpl-collect/collect-common.c	129	1
22	Contiki3.0/examples/ipv6/rpl-collect/collect-common.c	137	1
23	Contiki3.0/apps/shell/shell.c	247	1
24	Contiki3.0/apps/shell/shell.c	531	1
24	Contiki3.0/apps/shell/shell-download.c 	152	1
26	Contiki3.0/apps/shell/shell-sendtest.c	138	1
27	Contiki3.0/apps/shell/shell-sendtest.c	148	1
28	Contiki3.0/apps/shell/shell-time.c	248	1
29	Contiki3.0/apps/shell/shell-time.c	323	1
30	Contiki3.0/apps/shell/shell-netperf.c	576	1
31	Contiki3.0/apps/irc/ircc.c	216	1
32	Contiki3.0/apps/webbrowser/www.c	309	2
33	Contiki3.0/apps/email/email.c	188	3
34	Contiki3.0/cpu/6502/ipconfig/ipconfig.c	132	3
35	Contiki3.0/apps/netconf/netconf.c	128	3
36	Contiki3.0/apps/telnet/simpletelnet.c	160	3
37	Contiki3.0/tools/z80/hex2bin/src/hexameter.c	95	4
38	Contiki3.0/examples/ipv6/slip-radio/no-framer.c	70	5
39	Contiki3.0/core/net/mac/framer-802154.c	82	5
40	Contiki3.0/core/net/mac/sicslowmac/sicslowmac.c	89	5

2. memcpy API may introduce memory overflow, like the bug in "https://github.com/contiki-os/contiki/issues/707". In the calling process of memcpy,it may cause overflow when the third parameter is a constant, but the the second parameter is a char * type. Because memcpy will not stop when it encounter a ‘\0’, but the length of the second parameter may less than the value of the third parameter. The followings are potential bugs of this type in contiki2.7 and contiki3.0.
1)	contiki-2.7/apps/antelope/index-maxheap.c, line 516 
memcpy(index->descriptor_file,filename,sizeof(index->descriptor_file))， 
2)	contiki-2.7/apps/antelope/index-maxheap.c , line 539
memcpy(bucket_filename,filename,sizeof(buket_filename) 
3)	contiki-2.7/apps/antelope/relation.c, line 247
memcpy(rel->name,name,sizeof(rel->name))，sizeof(rel->name) 
4)	ontiki-2.7/core/cfs/cfs-coffee.c, line 654
memcpy(hdr.name,name,sizeof(hdr.name))，sizeof(hdr.name)
5)	contiki-2.7/core/net/rime/broadcast-announcement.c, line 139
memcpy(&adata,ptr,sizeof(struct announcement_msg))
6)	contiki-2.7/core/net/rime/broadcast-announcement.c, line 154
memcpy(&data,ptr,sizeof(struct announcement_data)) 
7)	contiki-2.7/core/net/rime/polite-announcement.c, line 134
memcpy(&adata,ptr,sizeof(struct announcement_msg))
8)	contiki-2.7/core/net/rime/ polite-announcement.c, line 149
memcpy(&data,ptr,sizeof(struct announcement_data)) sizeof(struct announcement_data)
9)	contiki-2.7/examples/ipv6/rpl-collect/collect-common.c, line 99
memcpy(&data,payload,sizeof(data))
10)	contiki-3.0/apps/antelope/index-maxheap.c, line 516
memcpy(index->descriptor_file,filename,sizeof(index->descriptor_file))，sizeof(index->descriptor_file) 
11)	contiki-3.0/apps/antelope/index-maxheap.c, line 539
memcpy(bucket_filename,filename,sizeof(buket_filename))，sizeof(buket_filename) 
12)	contiki-3.0/apps/antelope/relation.c, line 247
memcpy(rel->name,name,sizeof(rel->name))，sizeof(rel->name)
13)	contiki-3.0/core/lib/aes-128.c, line 90
memcpy(round_key[0],key,AES_128_KEY_LENGTH)，AES_128_KEY_LENGTH
14)	contiki-3.0/core/net/llsec/ccm-star/c, line 62  
memcpy(nonce+1,extended_source_address,8) 
15)	contiki-3.0/core/net/rime/broadcast-announcement.c, line 137
memcpy(&adata,ptr,sizeof(struct announcement_msg)) 
16)	contiki-3.0/core/net/rime/broadcast-announcement.c, line 157
memcpy(&data,ptr,sizeof(struct announcement_data))
17)	contiki-3.0/core/net/rime/polite-announcement.c, line 132
memcpy(&adata,ptr,sizeof(struct announcement_msg)) 
18)	contiki-3.0/core/net/rime/ polite-announcement.c, line 147
memcpy(&data,ptr,sizeof(struct announcement_data)) 
19)	contiki-3.0/dev/enc28j60/enc2j60.c, line 441
memcpy(enc_mac_addr,mac_addr,6) 
20)	contiki-3.0/examples/ipv6/rpl-collect/collect-common.c, line 99
memcpy(&data,payload,sizeof(data)) 
21)	contiki-3.0/Core/net/llsec/ccm-star-packetbuf.c, line 23
memcpy(nonce,extended_source_address,8) 
22)	contiki-3.0/Core/net/llsec/ccm-star-packetbuf.c, line 43
memcpy(nonce,extended_source_address,8) 
3."sprintf" API used inappropriately may cause system performance loss. at the locations in the following ：
1)        contiki-2.7/apps/shell/shell-vars.c,line 108
2)        contiki-3.0/apps/shell/shell-vars.c, line 108
the sprintf API is used to transfer format  of characters,but this ways very inefficiently. I have a experiment on sprintf API that transfer 64 characters to "%02x" format. the cost time is 159138ns, but the cost time is 90806ns when use a special arithmetic. when transfer 16 character to "%02x" format, the special arthimetic is:
 static char hexmap[]= { '0', '1', '2', '3', '4', '5', '6', '7', '8', '9', 'a', 'b', 'c', 'd', 'e', 'f' };
    int i;
    char *p;

    p=(char *) str->ptr();
    for (i=0;i<=15;i++)
    {
      (*p++)=hexmap[digest[i] >> 4];
      (*p++)=hexmap[digest[i] & 15];
    }
    str->length((uint) 32);
    return str;

zq301, 2016-01-18:
I hope anyone can help me check my finds as soon as possible. And if there are any unclear things in my statements, please let me know. Thank you!

tussock-cam, 2016-01-19:
I'm not sure why you think those `memcpy` calls are an issue? Just checking a random one, 6) is fine and presumably the others are too. `memcpy` is declared as `void *memcpy(void *str1, const void *str2, size_t n)`, the parameter names may be confusing you -- it copies bytes from an arbitrary address, it doesn't copy null terminated character arrays.

Regarding your `sprintf` comment, yes, custom code is probably faster but it is less general and harder to adapt.

zq301, 2016-01-19:
@tussock-cam  Thank you for your reply. Yes, memcpy can copy bytes from an arbitrary address, so it can copy from a null terminated character array, just like the bug in "https://github.com/contiki-os/contiki/issues/707". The third parameter of memcpy is a constant, so we must have a length check before every copy. Because memcpy will not stop copy(even encounter a null character) till the length required.

About the sprintf issue, I don't know the use frequency of that codes. But if the codes used very frequently, I think it is better to fix it.  It is not general but it will save much CPU time after fixing. We can fix them one by one,  CPU time is very valuable in IOT after all. The fix is very simple and I don't why it is harder to adapt?