contiki-os/contiki #1458: Debugging TI sensortag web-demo with CC-devpack debugger rev 1.2.0 gives error: could not open flash to load config

weijian19391, 2016-01-12:
Hi! 

I'm facing the error of unable to open ext flash while debugging with the Devpack debugger. I understand previously when debugging with srf-06 the error occurs because we are unable to use UART with ext flash, may i know if this is still the case for the Devpack debugger?

g-oikonomou, 2016-01-13:
I cannot remember off the top of my head, but have you tried a full powercycle of your tag after programming it?

weijian19391, 2016-01-13:
Hi! 

May I know how do I do that??

g-oikonomou, 2016-04-04:
Does this happen all the time or does it only happen after programming the tag? Have you tried to fully powercycle it?

I suspect something on the devpack may be messing around with SPI pins, in a fashion similar to #1154

weijian19391, 2016-04-04:
Hi @g-oikonomou 
It happends all the time even after I powercycle the sensortag, not sure what's causing the issue though.

g-oikonomou, 2016-04-04:
#1581 
and
http://forum.43oh.com/topic/8822-warning-jtag-interferes-with-sensortag-external-flash-access/

closed 2016-04-04 by g-oikonomou