contiki-os/contiki #1460: Sensortag web-demo MQTT debugging

anatoly-kr, 2016-01-13:
I'm looking for advice on how to debug mqtt connection from sensortag with "cc26xx-web-demo" connected to mesh network through Raspberry PI + devpack <--> Sensortag (CC2650STK) "slip-radio" acting as 6Lowpan Border Router (6LBR). "cc26xx-web-demo" was built with CC26XX_WEB_DEMO_CONF_MQTT_CLIENT 1 in project-conf.h

Sensortag is accessible through IPv6 - ping, coap and http from laptop in the same subnet with Raspberry 6LBR. 

Raspberry PI has Mosquitto MQTT broker running and accessible through IPv4 and IPv6 from Ethernet, however I cannot get Sensortag web-demo to publish messages to Raspberry MQTT IPv6 address.

I've changed default IBM Quickstart Broker IP on MQTT/IBM Cloud Config page to IPv6 address of Raspberry 6LBR and through tcpdump on Raspberry 6LBR I cannot see any packets from SensorTag trying to reach MQTT 1883 port on Raspberry, though I see HTTP traffic floating from my laptop and back to SensorTag and I can see test MQTT traffic from laptop to Raspberry over IPv6.

After changing MQTT config SensorTag switches LED off after few minutes of slow blinking .

I would appreciate any advice on how to start debugging if MQTT is running on SensorTag or it fails for some reason.

************* Raspberry 6LBR **************
root@raspberrypi:/home/pi# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP qlen 1000
    link/ether b8:27:eb:35:99:9e brd ff:ff:ff:ff:ff:ff
3: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP
    link/ether b8:27:eb:35:99:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.40/24 brd 192.168.1.255 scope global br0
       valid_lft forever preferred_lft forever
    inet6 bbbb::ba27:ebff:fe35:999e/64 scope global dynamic
       valid_lft 86323sec preferred_lft 14323sec
    inet6 fe80::ba27:ebff:fe35:999e/64 scope link
       valid_lft forever preferred_lft forever
4: tap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UNKNOWN qlen 500
    link/ether 02:0a:0b:0c:0d:0e brd ff:ff:ff:ff:ff:ff
root@raspberrypi:/home/pi#  route -A inet6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
bbbb::/64                      ::                         UAe  256 0     0 br0
fe80::/64                      ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1     9 lo
::1/128                        ::                         Un   0   1     3 lo
bbbb::ba27:ebff:fe35:999e/128  ::                         Un   0   1     0 lo
fe80::ba27:ebff:fe35:999e/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 1     0 br0
ff00::/8                       ::                         U    256 0     0 tap0
::/0                           ::                         !n   -1  1     9 lo

******** 6lbr.conf ************
MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0

**********************************
SensorTag responds to ping aaaa::212:4b00:796:a407

g-oikonomou, 2016-01-13:
Very likely to be down to 6lbr configuration. Try a vanilla border-router on a tag+devpack and run tunslip6 on your RPi and it should work.

anatoly-kr, 2016-01-13:
Thanks for quick reply, will try and report

g-oikonomou, 2016-01-13:
Just to clarify my experience: I have 6lbr running on BeagleBone Black + some slip radio (usually CC2531 USB dongle. The BBB cannot talk to devices in the mesh, but other PCs in my network can talk to the mesh through the BBB. Perhaps this is what you are seeing.

anatoly-kr, 2016-01-13:
Yes, exactly that, I can access mesh from laptop, but not from Raspberry
I hoped to get Raspberry to act as router + mqtt, sounds it's not an option anymore

g-oikonomou, 2016-01-13:
I'm sure you can do this with a different 6lbr configuration, but I cannot help any further nor is this the correct forum for this discussion. Alternatively, run MQTT on a different box if you have one :)

anatoly-kr, 2016-01-13:
Sure, thanks. Maybe, any thoughts on how to see if I could see DBG() results from mqtt-client.c through network PC over mesh?

g-oikonomou, 2016-01-13:
Nothing remotely that I have used before. Attach your mesh node to a device that can display its UART output until you are ready to run silent. That's what I normally do anyway.

anatoly-kr, 2016-01-13:
Ok, thanks.

closed 2016-01-13 by anatoly-kr