contiki-os/contiki #1461: COAP_BLOCKING_REQUEST asumes piggybacked CoAP request.

ezequiel-umu, 2016-01-15:
Using erbium + coap, I've seen in examples that if you use

```C
COAP_BLOCKING_REQUEST(&server_ipaddr, REMOTE_PORT, request, client_chunk_handler);
```

The function `client_chunk_handler` gets called at the moment of receiving the first ACK. The problem comes if the server's response comes separated, and not piggybacked into this ACK. 

Is there any way to block until request even if it comes separated?

mguc, 2016-01-15:
Hello Ezequiel

You are right, the er-coap-engine is just working for piggybacked
responses. I did a change in order to allow responses coming in separate
messages (Checkout my FIX
<https://gist.github.com/mguc/6cdc63a1efae479b2f62> here). Following the
standard [rfc7252 <https://tools.ietf.org/html/rfc7252>] this should be
supported at a certain point by Contiki. The fix is not pretty and I'm not
proud of it but it works for my case.

Best regards
Marko

ezequiel-umu, 2016-01-16:
Could you attach an example of use? It would be very useful.

mguc, 2016-01-22:
I'm working on a project where we use a node CoAP implementation ([node-coap]https://github.com/mcollina/node-coap) on the border router side and the Contiki CoAP implementation on the client side. That's why I can't provide a full example. If I send a COAP_BLOCKING_REQUEST to the CoAP-Server, the answer might come in the ACK packet or in a separate CON message with a new message id, depending on if the response is immediately ready or not. The trick is, as long as there is no payload in the response (which is identified by the message id of the first sent packet from the client) I keep the transaction active. Meanwhile the "real" response with the payload arrives (but a different message id). At the moment the er-coap-engine can't associate this response to the original request (and message id). That's why I memorize the current transaction. In my case, I have not more than one COAP_BLOCKING_REQUEST running simultaneously (not sure if this is even possible) and therefore I know that the incoming CON packet belongs to the request. As I said, this is just a dirty fix and not the proper way to do it. The best way would be to handle requests with a token id. Sorry for the late reply.

cladmi, 2016-12-15:
I have the same problem here.
I'm working with a contiki server with a resource that is made like `res-b1-sep-b2.c` except returning static data.

cladmi, 2016-12-15:
And your current fix gets the beginning of my data corrupted for the first 3 bytes.

It happens in this line https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-ext-header.c#L563
Which is called by https://github.com/contiki-os/contiki/blob/master/apps/er-coap/er-coap-engine.c#L272

akindestam, 2019-03-27:
This is still a problem in 2019.
Is there any workaround? Will the callback API be able to handle this?