contiki-os/contiki #1465: CC1310DK keep rebooting runing cc26xx-demo with latest Contiki version.

chenek, 2016-01-18:
Just sync to latest Contiki version and build cc26xx-demo for CC1310DK with "make TARGET=srf06-cc26xx BOARD=srf06/cc13xx cc26xx-demo.bin". The binary will make my CC1310 keep rebooting.

chenek, 2016-01-18:
arend had find the cause of this problem (https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/482914#pi239031350=2) and need someone to verify this. 

found in "Merge pull request #1350 from g-oikonomou/bugfix/cc26xx/tx-power" the reason.

line 163: *status = HWREG(RFC_DBELL_BASE + RFC_DBELL_O_CMDSTA) & 0xFF;

if you get rid of  masking "& 0xFF" everything works fine again.

g-oikonomou, 2016-01-18:
Any idea why? The E2E thread doesn't discuss the "why" either and I cannot see any logic in that. Are we certain this is a proper fix and not an accidental one?

g-oikonomou, 2016-01-18:
Also, any idea exactly which command cause this?

chenek, 2016-01-18:
After I build and download binary into my CC1310DK, it just keeps reboot. There is no command to trigger this.

g-oikonomou, 2016-01-18:
I mean the code tries to send a command to the RF and the code gets stuck there waiting forever. I am wondering whether it's the first command or if it's a particular one.

Thanks for reporting this!

g-oikonomou, 2016-01-24:
@chenek please, can you check with #1479?

chenek, 2016-01-24:
I don't have access to my CC1310DK. I will test this on Tuesday and report back to you. Thanks for notifying us about this patch.

chenek, 2016-01-27:
I have tested on this patch and it doesn't reboot now

PR #1479 (2016-01-27): Allow the caller to access the entire content of CMDSTA
When sending a command to the CC13xx/CC25xx RF core, we wait for command completion by checking the LSB of CMDSTA (correctly so). However, in doing so we also zero out the 3 CMDSTA return bytes. For some commands, those bytes contain useful information (e.g. an RSSI value) and are required by the caller.

This problem manifests itself e.g. in PROP mode `channel_clear()`, whereby the caller will always see an RSSI value of 0.

This pull therefore fixes the logic in `rf_core_send_cmd()` to check for command completion by blocking on the CMD [... cut]
commit d42b1b50 (2016-01-27): Allow the caller to access the entire content of CMDSTA

When sending a command to the CC13xx/CC25xx RF core, we wait for command completion by checking the LSB of CMDSTA (correctly so). However, in doing so we also zero out the 3 CMDSTA return bytes. For some commands, those bytes contain useful information (e.g. an RSSI value) and are required by the caller.

This problem manifests itself e.g. in PROP mode `channel_clear()`, whereby the caller will always see an RSSI value of 0.

This pull therefore fixes the logic in `rf_core_send_cmd()` to check for command completion by blocking on the CM [... cut]
closed 2016-01-27 by chenek