contiki-os/contiki #1482: SensorTag unable to connect to 6lbr

neoengtai, 2016-01-26:
I have built and run the demo application in `examples/cc26xx/cc26xx-web-demo` but the SensorTag does not seem to be able to connect to a 6lbr running on RaspberryPi. The 6lbr web console does not show any connected device. 

However, I managed to get the demo to work by checking out the commit f86494c6aa43375b0c9110294e4bf3b89a1c633f in `core/net/ipv6`. The commits thereafter have made changes to `sicslowpan.c` which I do not know if it's the main cause of the problem. Am I missing out on something?

The 6lbr version I am using is 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49).

msolters, 2016-01-28:
FWIW mine works using the latest `develop` branch of 6LBR.  What does the output of `tail -f /var/log/6lbr.log` look like from your Pi?

neoengtai, 2016-02-02:
I have not tried it with the latest develop branch of 6LBR. But with the current version I'm using, with log level 4, there are only some SCMD lines.

closed 2016-02-20 by g-oikonomou