contiki-os/contiki #1499: Way to pause or stop RPL?

aldorianmode, 2016-02-03:
Hello everyone, 

As the title says, there is a way to pause or stop RPL? Im on a project that because of hardware limitations, can't use the radio and manage a device via UART at the same time. Trying to turn the radio off works for a little while but it seems that RPL turn it on again. 

I'd be gratefull for any idea or suggestion,

Aldo

matfu2, 2016-02-04:
The flag CONTIKI_WITH_RPL defaults is set to default to 1 in Makefile.include . 
To remove RPL, specify "CONTIKI_WITH_RPL=0" in the console when make-ing, or put "CONTIKI_WITH_RPL=0" in a project makefile.

aldorianmode, 2016-02-04:
Thanks for the quick answer @matfu2 :) maybe i wasn't entirely clear with my question but the thing is I need to pause and resume RPL on runtime. Is possible to do that?

matfu2, 2016-02-05:
Unfortunately, I can't find any built-in way to disable/re-enable RPL at runtime.

LanshunNie, 2016-03-03:
RPL is triggered by etimer. If you stop etimer, RPL stops.

aldorianmode, 2016-03-07:
Thanks for the help. For the purpose i needed to pause/stop RPL, its not necessary anymore. I could turn off the radio using NETSTACK_MAC.off(0) using contikimac. With nullrdc i couldn't do the same, must investigate why.

closed 2016-03-07 by aldorianmode

oddeyed, 2016-03-08:
Just curious - did you try NETSTACK_RDC.off(0)? Any good MAC layer should pass that down though.

hafss, 2017-06-01:
Hello, i have changed the RDC driver to contikimac, but i have always nullrdc in mote outeput window of cooja simulator:
 **671613:1:nullsec CSMA nullrdc, channel check rate 128 Hz, radio channel 26, CCA threshold -45**
Should i turn off the radio using NETSTACK_RDC.off(0)?