contiki-os/contiki #1517: TSCH on wismote - EB synchronization fail

anabildea, 2016-02-15:
Dear all,

I am trying  to port TSCH on wismote. I have cloned contiki3.0 from https://github.com/contiki-os/contiki.git.

At present, I have successfully compiled the wismote platform  with Orchestra and TSCH mac.

The problem is that I see that nodes associate at first EB received on broadcast  and then leave the network  due to  synchronization issue.
For example I get the error: ??!dl-miss TxBeforeTx? on the coordinator .
Checking  the file hereafter:

./core/net/mac/tsch/tsch-slot-operation.c:          TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX, "TxBeforeTx");

I suspect that my wismote timers are wrong set:

/* Delay between GO signal and SFD: radio fixed delay + 4Bytes preample + 1B SFD -- 1Byte time is 32us
 * ~327us + 129preample = 456 us */
#define RADIO_DELAY_BEFORE_TX ((unsigned)US_TO_RTIMERTICKS(192)) //456
/* Delay between GO signal and start listening
 * ~50us delay + 129preample + ?? = 183 us */
//#define RADIO_DELAY_BEFORE_RX ((unsigned)US_TO_RTIMERTICKS(183))
/* Delay between the SFD finishes arriving and it is detected in software */
//#define RADIO_DELAY_BEFORE_DETECT 0

Do you know  what?s the procedure to compute the right TX/RX delay  timers?

Second, on the node side, the node associates but doesn't synchronize at all:

00:15.571        ID:2    TSCH: {asn-0.4c9 link-1-17-1-1 ch-26} !dl-miss association -31492 5248
00:15.581        ID:2    TSCH: {asn-0.4ca link-1-17-2-1 ch-20} !dl-miss association 28815 328
00:15.591        ID:2    TSCH: {asn-0.4d8 link-2-31-0-2 ch-26} !dl-miss association 28505 4592
00:15.600        ID:2    TSCH: {asn-0.4da link-1-17-1-1 ch-20} !dl-miss association 23933 656
00:26.443        ID:2    TSCH: send packet to 1 with seqno 1, queue 0 1, len 21 21
00:26.446        ID:2    TSCH: sending KA to 1
00:26.574        ID:2    TSCH: {asn-0.96f link-1-17-1-1 ch-15} !dl-miss TxBeforeAck 126 118
00:26.582        ID:2    TSCH: {asn-0.96f link-1-17-1-1 ch-15} uc-1-0 21 tx 1, st 2-1
00:26.914        ID:2    TSCH: {asn-0.991 link-1-17-1-1 ch-26} !dl-miss TxBeforeAck 126 118
00:26.922        ID:2    TSCH: {asn-0.991 link-1-17-1-1 ch-26} uc-1-0 21 tx 1, st 2-2
00:27.254        ID:2    TSCH: {asn-0.9b3 link-1-17-1-1 ch-15} !dl-miss TxBeforeAck 126 118
00:27.263        ID:2    TSCH: {asn-0.9b3 link-1-17-1-1 ch-15} uc-1-0 21 tx 1, st 2-3
00:28.616        ID:2    TSCH: {asn-0.a3b link-1-17-1-1 ch-15} !dl-miss TxBeforeAck 126 118
00:28.624        ID:2    TSCH: {asn-0.a3b link-1-17-1-1 ch-15} uc-1-0 21 tx 1, st 2-4
00:29.807        ID:2    TSCH: {asn-0.ab2 link-1-17-1-1 ch-20} !dl-miss TxBeforeAck 126 118
00:29.815        ID:2    TSCH: {asn-0.ab2 link-1-17-1-1 ch-20} uc-1-0 21 tx 1, st 2-5
00:35.592        ID:2    TSCH: {asn-0.cf4 link-1-17-1-1 ch-25} !dl-miss TxBeforeAck 126 118
00:35.601        ID:2    TSCH: {asn-0.cf4 link-1-17-1-1 ch-25} uc-1-0 21 tx 1, st 2-6
00:48.866        ID:2    TSCH: {asn-0.1222 link-1-17-1-1 ch-20} !dl-miss TxBeforeAck 126 118
00:48.874        ID:2    TSCH: {asn-0.1222 link-1-17-1-1 ch-20} uc-1-0 21 tx 1, st 2-7
00:49.376        ID:2    TSCH: {asn-0.1255 link-1-17-1-1 ch-26} !dl-miss TxBeforeAck 126 118
00:49.385        ID:2    TSCH: {asn-0.1255 link-1-17-1-1 ch-26} uc-1-0 21 tx 1, st 2-8
00:55.494        ID:2    TSCH: KA sent to 1, st 2-9
00:55.506        ID:2    TSCH: {asn-0.14b9 link-1-17-1-1 ch-26} !dl-miss TxBeforeAck 126 118
00:55.515        ID:2    TSCH: {asn-0.14b9 link-1-17-1-1 ch-26} uc-1-0 21 tx 1, st 2-9
01:01.373        ID:2    TSCH: send packet to 255 with seqno 2, queue 0 1, len 15 62
01:01.445        ID:2    TSCH: leaving the network
01:01.456        ID:2    TSCH: {asn-0.170c link-1-17-1-1 ch-25} ! leaving the network, last sync 4708
01:01.460        ID:2    TSCH: update time source: 1 -> 0
01:01.465        ID:2    TSCH-schedule: remove_link 0 2 1 0 0
01:01.470        ID:2    TSCH-schedule: add_link 0 2 2 0 0 0
01:01.475        ID:2    TSCH: scanning on channel 26
01:40.321        ID:2    TSCH: scanning on channel 25
01:40.677        ID:2    TSCH: association: received packet (95 bytes) on channel 25
01:40.682        ID:2    TSCH:! parse_eb: no join priority

Any ideas what I am missing?

Thanks a lot.

Best regards,
Ana Bildea

PhilipTtz, 2016-02-23:
Hello Ana,

I am trying to port TSCH to the cc2538 and I am experiencing the same issues. The nodes associate but dont seem to synchronize. After some seconds the node leaves the network with 

TSCH;! parse_eb: frame is not a valid TSCH beacon.
TSCH:! failed to parse EB (len95)

I think that timer related problems might be the reason. Did you manage to fix your problems in the meantime? I would be glad to hear how you did that, maybe I could adapt your solution :)

Thanks a lot & have a great day,
Phil

simonduq, 2016-02-23:
Hi,

I didn't not reply to @anabildea as we had offline discussions before I suggested she opens this issue.

@PhilipTtz, yes, getting nodes to keep in sync is generally the hardest step when porting TSCH to a new platform. Please note that after nodes lose synchronization, the output you get ("not a valid beacon" etc) is perfectly valid: the node initiates a new scan, hears not only EBs but also DIOs or any other traffic, and will output the two lines above.

As for getting timing right, the two main causes I can think of:
(1) Wrong timestamping: maybe due to a wrong timer conversion (check the usec to/from ticks macros, and check the timebase of your radio timestamps) or a mis-configuration of the chip, the scanning node is going to start with a time shift and never hear anything else. Will leave the network after `TSCH_DESYNC_THRESHOLD`. The reliable way to troubleshoot this is to use a logic analyzer and check the alignement of both nodes's timeslot and/or tx and rx time.
(2) Nodes are synchronized but fail to communicate within a slot. You could try to add some low-level debugs, or better, use a logic analyzer, to figure out if the node is receiving at all. If it is indeed receiving, try to locate where it is dropped. Add more `TSCH_LOG_ADD` in `tsch-slot-operation` for non-intrusive logging during timeslot execution.

Other advice: to start with, set all `DELAY_BEFORE_` macros to 0. Will be suboptimal but will work with the (very large) default TSCH guard time of 2.2ms. Later, use a logic analyzer to measure the true values.

Something worth trying: use a slotframe length of 1 (`TSCH_SCHEDULE_DEFAULT_LENGTH`)  and set up TSCH on a single channel (`TSCH_HOPPING_SEQUENCE_1_1`). After joining, even if timestamping is wrong, nodes will hear each other out of luck every now and then (they will be listening for 2.2ms every 10ms, always on the same channel). Should help finding out if the problem is (1) or (2).

Simon

PhilipTtz, 2016-02-24:
Hello Simon,

thank you for your answer. 
I checked the macros and the porting again and can not find anything wrong. What I guess is happening, is that the system at some point is stuck in a routine that takes too long, so that I get the dl-miss association errors. The difference between rtimer now and the reference is at max 180, but at some point it jumps to over 1000.

My guess is some ISR that may take too long. 

Have a great day,
Philip

simonduq, 2016-02-24:
Hi,

If you have a series of `dl-miss` right after associating the cause is likely not a too long ISR, but rather wrong timestamping of the EB. After receiving an EB and deciding to join on it, a node will use the SFD timestamp and the schedule to calculate the next active slot. It will schedule a wakeup for this slot. It might realize, by the time it actually sets the rtimer, that the deadline is past; in this case, it iterates and schedules the next active slot. This normally happens 0 or 1 time. If you see many `dl-miss` then it means the timestamp was completely off. (I think on every platform I ported TSCH to, I had this problem at some point and fixing SFD timestamping sorted it out :))

Hope this helps,
Simon

anabildea, 2016-02-25:
Hi Philip, 

I have encountered the same behavior. And I suspect a SFD timestamping issue, as Simon pointed out too. Unfortunately ,I didn't have enough time to go further into details as I work on another projet at the moment.  I will let you know as soon as I have updates on the matter.

Best, 
Ana.

alignan, 2016-02-25:
I was just to start on this for the `RE-Mote` (also a cc2538-based platform), I will update my findings, cheers!.

svranand, 2016-03-10:
Hi Ana,

I too faced problems porting tsch code on Wismote. But I managed to get the code working by making changes to cc2520.c and using larger slot frame lengths. Found that the lack of support for SFD is one of the primary reasons. This means NETSTACK_RADIO.receiving_packet() call of tsch-slot-operation.c doesn't work. What I did was to  use the native cc2520.c driver code for receiving incoming packets, and emulated the polled mode so that tsch code is happy. There are other changes too that may be hard to describe here. It is not in the form that can be submitted to be merged with the main code.  Please contact me if you wish to have a look at the code.

Anand

anabildea, 2016-03-15:
Hi Anand, 

Thank you for sharing. I dropped you a line yesterday on your official email address.

Best, 
Ana

svranand, 2016-03-15:
Hi Ana,

Sorry for the delay in responding. I saw your mail yesterday. I will 
share the
code with you very soon, in a couple of days time as I am currently 
updating it
for the latest version of tsch.

Regards
Anand

on your official email address. > > Best, > Ana > >  > You are 
receiving this because you commented. >
view it on GitHub: > 
https://github.com/contiki-os/contiki/issues/1517#issuecomment-196828976Web 
Bug from 
https://github.com/notifications/beacon/AL2JZQ1NxBtPxi4JvgMu-dCF_A3J8Yaeks5ptrnrgaJpZM4HaL1n.gif

simonduq, 2016-03-15:
Please do not hesitate to PR this once you're happy with the code!

svranand, 2016-03-15:
Absolutely Simon!

SoonSYJ, 2016-12-27:
Hi @anabildea @svranand @simonduq 
Sorry at first. I have a question about the software workflow of TSCH_SCHEDULE_AND_YIELD().
The macro will yield the process which has called it. But I can not understand how the process get the control back? 

Please give me some advice.
Thanks a lot.
Best regards

TayyabaZainab0807, 2019-02-05:
Hi, I am trying to run TSCH over a different radio other then ieee-mode.c. Node gets associated with the network but after 1 or 2 packets (sending and receiving is working fine) it gives me an error " ASN drift of length -x-, leaving network" and device gets disassociated.
I am not sure why I might be facing this issue. Is it because of timers? or what?

Currently my timestamp is capturing the start of the received packet. Is this what TSCH requires ?

TayyabaZainab0807, 2019-02-05:
Hi Anand, thanks for the response. Can you send me you finding (slides n report?

svranand, 2019-02-06:
Hi,

I actually sent the slides and report as attachments. Looks like these are
not delivered. Can you send me any of your other email addresses ?

Regards
Anand

TayyabaZainab0807, 2019-02-06:
> Hi, I actually sent the slides and report as attachments. Looks like these are not delivered. Can you send me any of your other email addresses ? Regards Anand
> […](#)

17030052@lums.edu,pk

TayyabaZainab0807, 2019-02-06:
I guess I am facing this issue because the tx-command that I am using for my radio does not append  Timestamp of transmitted frame (but tsch requires Timestamp of transmitted frame for sync) and because of that i was using rx-queue timestamp.

Is there a way that I can add this Timestamp of transmitted frame to my transmitted packet as my packet header does not contain this piece of information?

TayyabaZainab0807, 2019-02-06:
I am actually confused about the time synchronization...if timestamp on child node only captured the time when the packet was received in the rx queue...and time source node does not attach it's timestamp to the packet then how child node get its clock sync with respect to parent node by only using its local time?

TayyabaZainab0807, 2019-03-08:
> Hi,
>
> If you have a series of `dl-miss` right after associating the cause is likely not a too long ISR, but rather wrong timestamping of the EB. After receiving an EB and deciding to join on it, a [...]

How to fix SFD? I can get the timestamp indicating the start of the frame but as my Radio do not have SDF adjustment so I am using cc2650 driver code for this but yet it's not working.