contiki-os/contiki #1522: Bugs in line 778 of mqtt-client for cc26xx-web-demo example

weijian19391, 2016-02-20:
Hi 

I am referring to this line: https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-web-demo/mqtt-client.c#L778

whereby there seems to be a little bug. 

This line is found under the case MQTT_CLIENT_STATE_PUBLISHING, where it does a check for ````state == MQTT_CLIENT_STATE_CONNECTED````. Isn't this statement forever not true, since we will only come to this line under the case where ````state == MQTT_CLIENT_STATE_PUBLISHING````?

g-oikonomou, 2016-02-20:
Nope, this is valid code.

If we enter the state machine under `state == MQTT_CLIENT_STATE_CONNECTED` then the switch will run the previous case block, which (intentionally) does _not_ have a break. Therefore, under some circumstances, we will enter `case MQTT_CLIENT_STATE_PUBLISHING`, but the value of `state` will still be `MQTT_CLIENT_STATE_CONNECTED`

closed 2016-02-20 by g-oikonomou

weijian19391, 2016-02-21:
@g-oikonomou Oh, i totally missed that no missing break statement. Thank you so much!