contiki-os/contiki #1524: Toggle leds srf06-CC1310 via mqtt mosquitto

magnerep, 2016-02-22:
Hello,

I was trying to toggle leds from srf06 board with cc1310 via mqtt misquitto

I followed these informations: https://github.com/contiki-os/contiki/blob/287092db422e03b86bb5827cd12677905db46f93/examples/cc2538-common/mqtt-demo/README.md

I changed #define CC26XX_WEB_DEMO_DEFAULT_ORG_ID "lowepanz"

After flashing i can ping the node but it is not publish anything in iot-2/evt/status/fmt/json ( this worked when ord id was quickstart ) and i send 0 and 1 commands to topic iot-2/cmd/leds/fmt/json but nothing is happening.

g-oikonomou, 2016-02-22:
If ORG ID is not "quickstart" then the client expects to also be provided with an auth token. If the auth token is not provided, then the client will error out immediately (before even trying to connect to the broker).

You have the following ways forward:

* Use "quickstart" with mosquitto for your tests (no auth token needed).
* Configure the ORG ID you wish to use, not by changing the define but by using the HTTP interface. This allows you to also set an auth token.
* Change the code so that it will accept different ORG IDs without insisting on an auth token. This can probably be done very easily by fiddling with the if block near line 720 in `mqqt-demo.c`: https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-web-demo/mqtt-client.c#L720
* Add a new define for the auth token (e.g. `CC26XX_WEB_DEMO_DEFAULT_AUTH_TOKEN`) and initialise `conf->auth_token` to the value of that define by adding a line inside the `init_config()` function of the same file. Bear in mind that this could/may/will break quickstart functionality.

magnerep, 2016-02-22:
I followed:

> Change the code so that it will accept different ORG IDs without insisting on an auth token. This can probably be done very easily by fiddling with the if block near line 720 in mqqt-demo.c: https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-web-demo/mqtt-client.c#L720
1. defined `#define LOWEPANZE "lowepanze"`
2. changed `if(strncasecmp(conf->org_id, LOWPANZ, strlen(conf->org_id)) != 0) {`

closed 2016-02-22 by g-oikonomou

MarCast, 2017-10-27:
Hi, I would like to make the same project. I download the mosquitto on my raspberry, but when I type iot-2/cmd/leds/fmt/json, I have a error.