contiki-os/contiki #1528: CC2650 power use way too high

msolters, 2016-02-23:
I have followed this discussion, but I still have the same type of problem.

I am using the very-sleepy-demo on a CC2650 sensortag, as compiled from the very latest master branch of this repo.  However when I run this code on the sensortag with a 3.3VDC power supply, it draws between *6mA - 11mA* (!!!).

I can access its COAP resources successfully, and toggle the app to/from "very sleepy mode".  The average current consumption looks like this:

*  Very sleep mode:  hovers around 6.72mA
*  Normal mode:  hovers around 7.3mA

I have observed this on multiple sensortag boards.  I cannot even get below 1mA, let alone in the microamp range!  What could be wrong?

I have followed [this thread](https://github.com/contiki-os/contiki/issues/1305), but it seems this issue was resolved at that time with a PR or two that was accepted, which means I am already using that code.

bkozak-scanimetrics, 2016-02-23:
Hello,

I don't really expect it to fix your problems outright, but I would appreciate it if you could check what impact, if any, #1531 has on your power consumption.

When I get the chance I'll see if I can find any problems with the sleep mode power consumption of the CC2650 with the current master and report back.

bkozak-scanimetrics, 2016-02-23:
I can reproduce your excessive power consumption by programming the sensortag with a flash image which was compiled for the CC2650 evaluation module.

When compiling for the sensortag you need to set ```BOARD=sensortag/cc2650``` on the command line to make (you will need to do a ```make clean``` to get rid of your old configuration first).

Although the vast majority of your power consumption should disappear with this change I did notice a few peculiarities:
- The sensortag takes about 50% more power in very sleepy mode compared to the evaluation module (9uA compared to 6uA). 
- The when the sensortag is in normal mode it sleeps with a 3mA current draw to the evaluation module's 1mA (could be just from a difference in LED power costs?)

msolters, 2016-02-23:
No, `make clean` has no effect on the power consumption of the sensor tags.  I always do that before a build, and I even manually cleaned out any binaries or linker files just to make sure.  With a completely clean build (`make TARGET=srf06-cc26xx BOARD=sensortag/cc2650`) I still end up with a power consumption of 7.30mA.

msolters, 2016-02-23:
[image]

bkozak-scanimetrics, 2016-02-23:
Sorry for jumping the gun there, the power consumption with the evaluation module image was exactly as you described so I thought that would be it for sure.

In this case I can't reproduce your issue with my hardware and the currently released master for Contiki.

When I was helping to debug the last set of power related issues I created a test application at https://github.com/bkozak-scanimetrics/CC2650_lpm_test which you might find helpful. This test can be used to measure the current draw of a CC2650 in it's lowest power mode. When I run it I see a draw of around 2 to 3 uA.

bkozak-scanimetrics, 2016-02-23:
> I have also noticed that, in "shutdown" sleep mode, the 2650 module for the SRF06 board also goes to about 3mA. This is usually from something on the order of uA while in "deep sleep," which I find very confusing. You'd think, if anything, it ought to be less.

Could be that your jumpers are set wrong for power measurement. You should have all the jumpers on ```XDS100v3 BYPASS``` removed as well as the ```Enable UART``` jumper. There are also two pins on ```P407``` that absolutely need to be left with jumpers on (seems to be fine if you leave the whole header jumpered).

g-oikonomou, 2016-02-24:
I measured the CC2650EM only just a couple of days ago with latest master and it was drawing less than 1uA in shutdown.

We spotted a leak when ALS_OUT is not mounted on P404 (is that what you meant by P407, Billy?).
 
Make sure to disconnect the JTAG.

bkozak-scanimetrics, 2016-02-24:
> We spotted a leak when ALS_OUT is not mounted on P404 (is that what you meant by P407, Billy?).

Yet that was the one.

msolters, 2016-02-24:
Thanks for the info!  This has been extremely helpful.  I've got power management within reason now.

closed 2016-02-24 by msolters

arurke, 2016-02-25:
@msolters Hi Mark, a bit confused, was it the jumpers, the flashing/image, or #1531 that solved it? :) Also, can you tell us what current draw you are seeing in deep sleep now? I'm just wondering what to expect, 9 uA is mentioned in this thread, 17 uA in #1305.

msolters, 2016-03-08:
I'm so sorry,

To be clear flashing the correct image absolutely had an impact as an order of magnitude drop in current.

However, the current consumption never seems correct with the srf06 board, and I have tried all combinations specified here and a lot of various combinations of jumpers/power sources/headers etc.

The only way I got a physically consistent value was by milling out my own "shield" for the CC2650 EM module and connecting to each of the pins without the srf06 board at all, just my own electronics.

In deep sleep mode with ContikiMAC I am seeing an average consumption of about 150uA, with very very narrow mA-scale spikes, but it does appear to be responsive and maintain the mesh.

bkozak-scanimetrics, 2016-03-08:
I don't have any suggestion to help you at this time but I do want to say that I also see the narrow mA spikes during sleep (even though my overall measured average consumption is less than yours). I believe that this is the expected behaviour from my understanding of the "recharge" function of the CC2650 in sleep mode.