contiki-os/contiki #1533: [Zoul Remote] Upload failed: Timeout waiting for ACK/NACK after 'Get ChipID (0x28)'

PalmaITEM, 2016-02-25:
Hi,
I'm using Instant Contiki but I've cloned the contiki-os repo since I am following the tutorial "IoT in five days".

I have no problems with Z1 but with Remote I cannot upload any code, not even the demo. Heres the output:
```
make zoul-demo.upload
using saved target 'soul'
  CC        ../../../cpu/cc2538/./ieee-addr.c
  CC        ../../../cpu/cc2538/cc2538.lds
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        zoul-demo.c
  LD        zoul-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
python ../../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00202000 zoul-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Get ChipID (0x28)'
make: *** [zoul-demo.upload] Error 1
rm obj_zoul/startup-gcc.o zoul-demo.co
```
Since I had been able to upload code before, and after trying to understand the erro, I tried an older version of _cc2538-bsl.py_ (version 1.1) and I was able to upload the code. However, when I connect to the serial line to debug nothing really happens.

If I use the old Contiki repo that comes with the Instant Contiki VM, the _remote-demo_ works just fine.

I'm a beginner in Contiki but please let me know if I can help further.

alignan, 2016-02-25:
Have you tested the instructions from this link?

https://github.com/Zolertia/Resources/wiki/Program-the-Zolertia-platforms

PalmaITEM, 2016-02-25:
Hi,
Thank you for the pointer! I was missing intelhex.

g-oikonomou, 2016-02-25:
I find it very difficult to understand how lack of IntelHex could cause the error you were getting. Can we try to clarify this, since if it was indeed the cause then the BSL could have a bug.

PalmaITEM, 2016-02-25:
As I said I not very experienced with Contiki. I'll try to explain better what happened and hopefully it'll be helpful.

I had tried to install python-magic before but it raised some problems (I used apt-get instead of pip and I may have run into incompatibilities maybe). Since it looked like an optional thing I reverted to try without it.

By using an earlier version of BSL I could upload the demo code but it only worked with the old demo code. If I tried the old BSL manually (without the %.upload) with the new repo code it would upload but nothing seemed to happen afterwards.

I did reboot my VM before trying the pip install, but I don't want to believe it was that. After installing IntelHex and python-magic things seem to be working fine with the latest Contiki repo.

g-oikonomou, 2016-02-25:
OK so I can confirm that the BSL works fine without intelhex (python 2.7 as well as 3.4) for the CC2650. So either the BSL has a problem for the CC2538, or we have a bug in the zoul's makefiles somewhere, or you're doing something wrong.

Can you please help debug this? I don't have my Re-Motes with me.

* Undo all your local changes and revert to the latest version of Contiki on master.
* Uninstall intelhex and python-magic fully from your system. If you installed them with pip, then `pip uninstall python-magic` will do it for you (you may have to run this with sudo if you get an error). The same for intelhex. If you installed them from apt then uninstall them with apt.
* Then use the .upload target and paste the output here.

Much appreciated

PalmaITEM, 2016-02-25:
Sure, glad to help.

Step 1: git reset --hard
`HEAD is now at 93b9089 Merge pull request #1504 from bthebaudeau/travis-force-aapcs-toolchain-version`
Step 2: sudo pip uninstall python-magic intelhex
Step 3: cd examples/zolertia/zoul/ && make clean && make zoul-demo.upload
```
using saved target 'zoul'
mkdir obj_soul
  CC        ../../../cpu/cc2538/./ieee-addr.c
  ... (removed the compilation part)
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
python ../../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00202000 zoul-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
make: *** [zoul-demo.upload] Error 1
rm obj_zoul/startup-gcc.o zoul-demo.co
```
Just to double check I re-installed both python-magic and intelhex and it is working again.

g-oikonomou, 2016-02-25:
Can you paste the working output too please?

PalmaITEM, 2016-02-25:
Here it is:
```
user@instant-contiki:~/contiki/examples/zolertia/zoul$ make zoul-demo.upload
using saved target 'zoul'
  CC        ../../../cpu/cc2538/./ieee-addr.c
  CC        ../../../cpu/cc2538/cc2538.lds
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        zoul-demo.c
  LD        zoul-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
python ../../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00202000 zoul-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
Firmware file: Raw Binary
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 06:16:0F:C9:00:12:4B:00
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Write 8 bytes at 0x0027FFF8F00
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xb6be4608)
rm obj_zoul/startup-gcc.o zoul-demo.co
```

g-oikonomou, 2016-02-25:
Thank you. I'm re-opening this.

g-oikonomou, 2016-02-25:
@PalmaITEM please, can you also try _with_ python-magic installed and without intelhex? I think I may be onto something.

PalmaITEM, 2016-02-25:
It worked now but it didn't when I tried python-magic through apt.
```
make zoul-demo.upload
using saved target 'zoul'
  CC        ../../../cpu/cc2538/./ieee-addr.c
  CC        ../../../cpu/cc2538/cc2538.lds
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        zoul-demo.c
  LD        zoul-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
python ../../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00202000 zoul-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
Firmware file: Raw Binary
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 06:16:0F:C9:00:12:4B:00
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Write 8 bytes at 0x0027FFF8F00
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xb6be4608)
rm obj_zoul/startup-gcc.o zoul-demo.co
```
Let me know if I can help more.

PalmaITEM, 2016-02-25:
I'm not sure if this is helpful but python-magic through apt-get doesn't work:
```
make zoul-demo.upload
using saved target 'zoul'
  CC        ../../../cpu/cc2538/./ieee-addr.c
  CC        ../../../cpu/cc2538/cc2538.lds
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        zoul-demo.c
  LD        zoul-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
python ../../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00202000 zoul-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
ERROR: 'module' object has no attribute 'from_file'
make: *** [zoul-demo.upload] Error 1
rm obj_zoul/startup-gcc.o zoul-demo.co
```

g-oikonomou, 2016-02-25:
It does help, yes. I suspect this to be a timing issue. My money is in this line here:
https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py#L216

I suspect that given a sufficiently high number there, programming will work with both packages uninstalled.

alignan, 2016-02-25:
In native ubuntu for me fails when flashing without `python-magic`, and with the package installed it works.  On my VM it works for both cases, so perhaps timing is the issue here, as the RE-Mote has an ob-board PIC detecting the BSL sequence over DTR/CTS, then driving the CC2538.

PalmaITEM, 2016-02-25:
I've tried going up to 2 seconds and it still does not work.

g-oikonomou, 2016-02-25:
OK so perhaps we need to sleep after L219 too. @alignan at what point in the script does the PIC asser the CC2538's RESET line?

PalmaITEM, 2016-02-25:
New update, fixed it by swapping the sleep and the last setDTR. I changed the code like this:
```
   #time.sleep(0.5)  # Make sure the pin is still asserted when the cc2538
                           # comes out of reset. This fixes an issue where there
                           # wasn't enough delay here on Mac.
        self.sp.setDTR(0 if not dtr_active_high else 1)
        time.sleep(0.005)
```

g-oikonomou, 2016-02-25:
Fantastic. I am certain it's the addition of the sleep after, not the removal of the sleep before that fixes it.

Thanks for your help!

PalmaITEM, 2016-02-25:
I actually commented the first sleep. If it small enough it's ok, but too large and it stops working. Maybe because it "breaks" the boot-loader invoking sequence.

My pleasure thank for your help as well.

alignan, 2016-02-25:
Yes, confirmed, this works.

````
    def invoke_bootloader(self, dtr_active_high=False):
        # Use the DTR and RTS lines to control bootloader and the !RESET pin.
        # This can automatically invoke the bootloader without the user
        # having to toggle any pins.
        # DTR: connected to the bootloader pin
        # RTS: connected to !RESET
        self.sp.setDTR(1 if not dtr_active_high else 0)
        self.sp.setRTS(0)
        self.sp.setRTS(1)
        self.sp.setRTS(0)
        time.sleep(0.002)  # Make sure the pin is still asserted when the cc2538
                           # comes out of reset. This fixes an issue where there
                           # wasn't enough delay here on Mac.
        self.sp.setDTR(0 if not dtr_active_high else 1)
        time.sleep(0.005)
````

I will make a PR with your suggested fix, thanks!

closed 2016-02-25 by alignan

PalmaITEM, 2016-02-25:
Glad I could help.

sebbad, 2016-12-13:
Hi,

I am having the same error message on mac os. I checked the suggested fixes, but they do not seem to have any effect for me.

smlng, 2017-02-28:
Hi,

I'm having the issue, too. But only on macOS (Sierra) on my iMac, not on my Macbook.