contiki-os/contiki #1537: With new LLSEC framer, fragmented packets have the same frame counter

laurentderu, 2016-03-02:
With the new LLSEC framer, fragmented packets have the same frame counter and so all the fragments, except the first one, are dropped at receiver side.

The root cause is that in sicslowpan.c, the packetbuf is not recreated between each fragment, but rather reused, and so the PACKETBUF_ATTR_SECURITY_LEVEL attribute is still set for the next fragment and add_security_header() does not update the frame counter.

Solution is either to explicitely reset the PACKETBUF_ATTR_SECURITY_LEVEL attribute between fragments, or to remove the condition in add_security_header(). @kkrentz, what do you think is the best solution ?

kkrentz, 2016-03-02:
Thanks for pointing this out. As long as we have the LLSEC layer, we can remove the condition and call add_security_header() from within LLSEC.send. Actually, this is how it was done beforehand ...

nfi, 2016-03-02:
Does not sicslowpan.c already recreate the packetbuf between each fragment? It stores the packetbuf to a queuebuf before transmission and then restores the packetbuf from the queuebuf after transmission. Unless PACKETBUF_ATTR_SECURITY_LEVEL is set before the first fragment is transmitted, the attribute should be cleared before transmission of each fragment. What am I missing?

laurentderu, 2016-03-02:
I'm not too familiar with queuebuf and packetbuf, for what I see I think the attributes are not recreated before each fragments, instead they are copied back from the queuebuf to the packetbuf after each call to packet_send() by calling queuebuf_to_packetbuf()

nfi, 2016-03-02:
Yes and that should restore the packetbuf attributes as setup by sicslowpan.c before sending the first fragment. No attributes set or any other packetbuf changes made during the transmission of a fragment will remain in packetbuf. queuebuf_to_packetbuf() overwrites everything in packetbuf including all attributes.

laurentderu, 2016-03-02:
Indeed, adding more traces shows that the PACKETBUF_ATTR_SECURITY_LEVEL attribute is set before we send the first packet. It is in fact the call to NETSTACK_FRAMER.length() that adds the security headers and the attribute.

This does not change the fix, but now the real root cause is known :)

nfi, 2016-03-02:
Ah, thank you! Then I understand the problem. :smiley: