contiki-os/contiki #1546: Porting contiki-os

harinaban, 2016-03-09:
Hello,
I am new to this community, I am working with a project that deals with contiki.

Now I am struggling with one point, I dont how to port contiki OS into real hardware (i.e) I am using 
MSP430F5529 launch pad in that I want to port the contiki.

I dont know the steps to port the contiki-os into it.

Please guys help me with this .

It will be grateful.

Thanks in advance.

~Hari

alignan, 2016-03-09:
You can take a look at how the platforms are implemented, check out the msp430-based platforms in the `platforms` directory.  There is also a thread on the contiki mailing list:

https://sourceforge.net/p/contiki/mailman/message/6339448/

Stack Overflow:

http://stackoverflow.com/questions/25232934/contiki-port-to-msp430f5-launchpad

There is also this presentation:

https://prezi.com/nfc5osxlvqwt/porting-contiki-os-to-discovery-board/

Exactly how are you struggling? what have you tried to do and how are you failing? have you googled and found this information already?

harinaban, 2016-03-17:
Hi Alignan,

sorry for the delay, 

I tried remaking a platform similar to wismotein platform folder, I named my platform as vmote and changed the following file **Makefile.vmote** and created a folder in example folder as **vmoteapp** and added a simple blink-hello.c file and compiled it using the following command by connecting my launchpad via USB.

 ->  vec@vec-desktop:~/contiki-2.7/examples/vmoteapp$ make
 ->  vec@vec-desktop:~/contiki-2.7/examples/vmoteapp$ make TARGET=vmote savetarget 
 ->  vec@vec-desktop:~/contiki-2.7/examples/vmoteapp$ make blink-hello.upload 
 
 At the end of the upload process I am getting the following error:

**msp430flasher -n msp430x5529 -e ERASE_MAIN -w blink-hello.hex -v -z [VCC]
/bin/sh: 1: msp430flasher: not found
make: *** [blink-hello.upload] Error 127
rm blink-hello.hex**

But I have installed the msp430-flasher tool in contiki-2.7 folder, Then too I am getting the same error message.

Please Help me .

Thanks in advance.

~Hari

alignan, 2016-04-24:
This is unrelated to the original question, but specific to your toolchain and development environment.  Try to add the location of msp430flasher to your PATH, else forward to Contiki mailing list.

closed 2016-04-24 by alignan