contiki-os/contiki #1554: CC2650 - Configuration for 24MHz XOSC only

adriencapaine, 2016-03-14:
Hello,

I developed a custom electronic board based on the CC2650. On this board, there is only one oscilllator (24MHz one) to drive the chip. Indeed, I have read on the manufacturer datasheet that only this oscillator was mandatory for the cc2650 to operate. But it looks like I should have included the LF one (32 KHz) as Contiki cc2650 libraries highly rely on Low Power Mode (which necessarily uses LF XOSC). :)

Therefore, I wondered if i could disable Low Power Mode entries to "force" all peripheral to remain in "always_on" domain. I was messing up the cc26xxWare drivers when I saw that the Low Clock Source (32K) could be derived from the HF oscillator (24M).
Lines 339 to 356 from file 'cc26wwware/inc/hw_ccfg.h' describe the mechanism. Unfortunately, cc2650 default setup configuration ('cc26xxware/driverlib/setup.c')make use of the LF oscillator. Which, in my case, halts my board as there is no such oscillator connected to my cc2650.

So here are my newbie questions:
Is is possible to modify this setting safely? Can I change the CCFG_BASE register (e.g. with HWREG(CCFG_BASE) = xxx) so that I always go for case 0 in setup.c?

I would gladly appreciate any help or advice before I do something really nasty about my board ;)
Kind regards

g-oikonomou, 2016-04-13:
According to the TRM, you can use the external 24MHz XOSC to feed the low-frequency clock. Alternatively, you could use the internal low-frequency RC OSC to feed the LF clock, but you should _really_ avoid doing so, because it's bugged: www.ti.com/lit/pdf/SWRZ058

Contiki assumes the presence of a low-frequency crystal and sets it as the source for the LF clock. You will have to modify this. I am not sure what changes you may have to make in xxware.

Keeping the 24MHz XOSC active all the time will have serious implications on energy consumption.

adriencapaine, 2016-05-09:
Thank you very much George for your reply
 I finally managed to get my custom board working by changing the oscillator settings of the _cc26xxware/startup_files/ccfg.c_. Arround line 110, we can change the clock options.
It's hard to say but (for now) I do not mind consuming as I am not battery-powered... Though my conscience takes a heavy hit for that ;)
I'll mark it as a solved issue

closed 2016-05-09 by adriencapaine