contiki-os/contiki #1556: Can't build er-rest-example for cooja

bsmelo, 2016-03-15:
Hello,

Sorry if I shouldn't open this here, but I could not find an existing similar issue, and this one is really blocking me.

I can't `$ make TARGET=cooja server-only.csc`. I've tried on:
1. InstantContiki3.0's folder [/home/user/contiki/](http://pastebin.com/tiH2W2jm), which I'm assuming is `release-2-7` branch
2. InstantContiki3.0's folder [/home/user/contiki-3.0/](http://pastebin.com/XJXKgDd7), which I'm assuming is `release-3-0` branch
3. Cloning `master` into InstantContiki3.0's folder [/home/user/master/](http://pastebin.com/vGDHsKtd)
4. Cloning `master` into my machine's folder [/home/bruno/contiki/](http://pastebin.com/VKceXAb0c)

I've **pasted links to the kind of errors I'm getting in each situation**, but just to summarize:
First, all of them complained about `git submodule update --init` and I ran it; Then,
1. 2. and 3. throws stuff like:
```
FATAL [main] (Cooja.java:3080) - Error when trying to read JAR-file in /home/user/contiki/tools/cooja/apps/mspsim: java.io.FileNotFoundException: cooja_mspsim.jar
FATAL [main] (Cooja.java:445) - Error when loading extensions: Error when creating class loader
```
(and that file really isn't there).

And 4. Complains about insufficient memory, even though :
```
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: er-example-server.sky section `.text' will not fit in region `rom'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [000000000000ffe0,000000000000ffff] overlaps section .text loaded at [0000000000004000,00000000000101bf]
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom' overflowed by 3340 bytes
```

Could you please point me out on how should I proceed?

alignan, 2016-03-25:
Are you running the `git submodule update --init` in the Contiki root location?
You could always clone the `mspsim` at `tools`, but the first command should take care of this.

bsmelo, 2016-03-30:
HI alignan,

For 1., 2. and 3.: Originally not, but if I run it now it doesn't do or print anything.

Anyway, for 4., which is the case I'm focused at, since I got the furthest with it, I don't think the problem is related to `mspsim`, right? I get insufficient memory complaints, even after disabling RDC in (as per https://github.com/contiki-os/contiki/tree/master/examples/er-rest-example) `examples/ipv6/rpl-border-router`.

What else should I modify to adjust these memory settings?

alignan, 2016-03-30:
Sadly I think this is related to the stack size overgrowing the sky's ROM, the only thing would be to use less resources, see my answer below:

http://stackoverflow.com/questions/35894400/cannot-fit-er-example-server-to-telosb/36156708#36156708

The other alternative is to use another mote, you can reconfigure the simulation and use instead i.e the Z1 mote or alike.

bsmelo, 2016-03-30:
Hm, I see no issue in using another mote like Z1.

Could you point me to some instruction or file where I should make this change? I can't even open `server-only.csc` on Cooja :-(

alignan, 2016-03-30:
What you should do is open cooja:

`contiki/tools/cooja$ ant run`

Then on the menu open and reconfigure the `*.csc` file, it will guide you step by step, allowing you to select what type of mote to use while reconfiguring the example.

bsmelo, 2016-03-30:
I can't do `File > Open` for the `*.csc` in order to reconfigure, since if I try to open it I get:

[image]

alignan, 2016-03-30:
No, you aren't.

`File -> Open simulation -> Open and Reconfigure`

You should then see in a next step something like `Loading mote type`

I'm closing the issue as originally it was about sky and the `er-rest-example`.  If you have more questions, please address Contiki mailing list.

closed 2016-03-30 by alignan

pablocorbalan, 2016-03-30:
Also, @bsmelo you are using msp430-gcc 4.6.3 version. I think that upgrading to 4.7 may also reduce the size of the binary created. It might help you out to be able to run the example.

Sorry that I answered once the issue is closed!

aichaab, 2018-02-26:
Hello,
This is my first experience with contiki .I want to upgrade the MSP430-gcc version can any one help me to do it
thank you .