contiki-os/contiki #1559: Add REMote platform to Cooja

tiagodiogo, 2016-03-17:
hello,
from what i have been observing and experimenting, the amount of RAM and ROM needed to run a CoAP server with some link layer security and possibly DTLS is now very large for the space Z1 platform as to offer.
i believe it would be interesting and important to allow support on Cooja for REMote, which has better specs that Z1 and is already part of the Contiki supported hardware.
is this achievable? it would be centrainly be very helpfull for big stacks (COAP + DTLS + RPL + LLSEC)

thank you,
Tiago

joakimeriksson, 2016-03-17:
Everything is achievable!

Well - this is a bit non-trivial but we are (slowly) working on getting support for ARM / Cortex M3/4 based devices
in COOJA or at least for simulating Contiki devices. We have some initial work going on with the Emul8 emulator
and a separate radio-medium simulator that we so far successfully have send some traffic and also managed to
run some of the COOJA csc files in. It will not be as timing accurate as the COOJA / MSPSim combination from
start but it will improve over time and will provide equally good simulation I would say. We also aim to create a
more scalable simulation framework as COOJA stops at around 100 simulated / emulated nodes.

Best regards,
— Joakim Eriksson, SICS / Antmicro

tiagodiogo, 2016-03-18:
hello @joakimeriksson 

it's nice to hear that this is an issue already being worked on!
since integration with Cooja is still in progress, is the Emul8 currently suitable for experimenting?
i've looked at the project page but i did not fully understood if it could simulate communication among nodes with this tipe of stack and protocols.
i understand the virtual machines load a binary file, can that be the output of building, for example, a coap server for the remote platform?

kind regards,
Tiago

joakimeriksson, 2016-03-18:
Yes, it is!

https://github.com/emul8/emul8

Best regards,
— Joakim

closed 2016-04-24 by alignan