contiki-os/contiki #1560: Tunslip6 broken on mac os x

steffoon, 2016-03-18:
Hello,
I just found out that Commit c7bb3b7b9fb410b12d3a90775ac1f3ed010ec716 breaks the execution of tunslip6 on mac os x. 
Displayed error: tunslip6: main: open /dev/tun: Is a directory

Undoing the introduced changes of c7bb3b7b9fb410b12d3a90775ac1f3ed010ec716 fixes this issue.

Kind regards

simonduq, 2016-03-21:
@cladmi, any clue?

cladmi, 2016-03-23:
Ho, so it's broken on mac with my commit. I suspected it may happen.

Can you try reverting it, and replace "tun0" and "tap0" with "tun%d" and "tap%d" where I deleted code ?
I have no mac to try it myself.

Regards.

alignan, 2016-04-24:
@g-oikonomou could you try @cladmi suggestion?

saurabhbansal90, 2016-09-27:
Hello all, i am working on a embedded linux board and i want to use 6lowpan for getting data from sensors.

I am getting the data by 6lowpan into my board. On gateway i want to run tunslip6 to read the IPV6 data. But i am not able to run tunslip6 file, i tries the same in my PC in ubuntu as well but it is showing same problem as below :

 ./tunslip6 babe::1 -t tun0 -a localhost -p 2000
client: connect: Connection refused
tunslip6: can't connect to ``localhost:2000'': Connection refused

I ran below command also 
./tunslip6 aaaa::1/64 -s /dev/ttyO3 2>gar

and i got below response 
********SLIP started on ``/dev/ttyO3''                                          
tunslip6: main: open: No such file or directory

I think one file tun will be created in /dev and that file can be used to send data to serial terminal.
please let me know if some additional information is needed to identify or resolve the issue.

Thanks in advance,
Saurabh.

aignacio, 2016-09-27:
Hello @Saurabh,

The tunslip just create an emulated nic adapter for tunelling the IPv6 packets based on your serial slip device, did you check if the /dev/ttyOX device is sending packets via SLIP protocol? This /dev/tty* is busy? How did you compile the tunslip program?

aignacio, 2016-09-27:
Hello @Saurabh,

The tunslip just create an emulated nic adapter for tunelling the IPv6
packets based on your serial slip device, did you check if the /dev/ttyOX
device is sending packets via SLIP protocol? This /dev/tty* is busy?

saurabhbansal90, 2016-09-27:
Hello Mr. Aignacio, thanks a lot for this quick reply. I have attached the source for tunslip6 in this comment. I am cross-compiling this file for my board.
It seems like SLIP is getting started and the TUN0 is not being created.
And at the present moment there was no data on the /dev/ttyOX port.
[tunslip6.txt](https://github.com/contiki-os/contiki/files/496530/tunslip6.txt)

And along with that i have enabled **Tunneling**, **IPv6**, **Tun/TAP** in my **linux kernel 3.2**.
Please let me know if from the Kernel side anything else I have to enable to use this tool.

I hope this information gives you some clear picture for my problem, please let me know if some further information is needed.

greg-king5, 2016-09-29:
Are you running the command through `sudo`?

g-oikonomou, 2016-12-06:
> Can you try reverting it, and replace "tun0" and "tap0" with "tun%d" and "tap%d" where I deleted code ?

Doesn't help. There is also a problem when specifying `-s`, not sure whether it's the same commit that causes it.

```
$ sudo ./tunslip6 -s /dev/tty.Bluetooth-Incoming-Port fd00::1/64
********SLIP started on ``/dev/tty.Bluetooth-Incoming-Port''
tunslip6: main: open /dev/tun: Is a directory
```

vandaele, 2017-01-20:
> Can you try reverting it, and replace "tun0" and "tap0" with "tun%d" and "tap%d" where I deleted code ?

@cladmi Doesn't help me neither, while using in context of iot-lab with -a and -p options

```
$ sudo ./tunslip6 -L -t tun0 -a localhost -p 2000 "fd00::1/64"
slip connected to ``::1:2000''
tunslip6: main: open /dev/tun: No such file or directory
```

vandaele, 2017-04-27:
Just come back here to share the solution of my problem, that was just linked to my OS.
macOS Sierra doesn't come with tun/tap interface support by default.
I needed to install [TunTap](http://tuntaposx.sourceforge.net). Thus, a series of `/dev/tap*`and `/dev/tun*` are created.
Then my previous command launches successfully my tunslip.

thiagohd, 2017-04-27:
Thank you for sharing @vandaele, I was looking for a solution as well. I will test it tonight.

CamelShoko, 2021-07-28:
> Just come back here to share the solution of my problem, that was just linked to my OS.
> macOS Sierra doesn't come with tun/tap interface support by default.
> I needed to install [...]

it works, thanks.

13:18:23 SLIP started on [/dev/tty.SLAB_USBtoUART]
0000.017 opened tun device ``/dev/tun0''
0000.017 ifconfig tun0 inet6 mtu 1500 up
0000.027 ifconfig tun0 inet6 fd00:7878:7878::1/64 add
0000.033 sysctl -w net.inet6.ip6.forwarding=1
net.inet6.ip6.forwarding: 1 -> 1
0000.038 ifconfig tun0

tun0: flags=8851<UP,POINTOPOINT,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	inet6 fe80::1a65:90ff:fee0:6923%tun0 prefixlen 64 optimistic scopeid 0xf
	inet6 fd00:7878:7878::1 prefixlen 64 tentative
	nd6 options=201<PERFORMNUD,DAD>
	open (pid 21814)
0000.042 Packet from TUN of length 64 - write SLIP
0000 60 00 00 00 00 18 3a ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff e0 69 23 87 00 fe ba 00 00 00 00 fe 80 00 00 00 00 00 00 1a 65 90 ff fe e0 69 23
0000.047 Packet from SLIP of length 72 - write TUN
0000 6a 02 b5 e1 00 20 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 4c 00 31 3a 00 05 02 00 00 01 00 83 00 7f a9 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 4c 00 31
0000.924 Packet from TUN of length 64 - write SLIP
0000 60 00 00 00 00 18 3a ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 00 00 01 87 00 8c b4 00 00 00 00 fd 00 78 78 78 78 00 00 00 00 00 00 00 00 00 01
0000.929 Packet from SLIP of length 64 - write TUN
0000 60 02 b5 e1 00 18 3a ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 4c 00 31 87 00 98 ac 00 00 00 00 fe 80 00 00 00 00 00 00 02 80 e1 ff fe 4c 00 31
0001.034 Packet from TUN of length 116 - write SLIP
0000 60 00 00 00 00 4c 00 01 fe 80 00 00 00 00 00 00 1a 65 90 ff fe e0 69 23 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 16 3a 00 01 00 05 02 00 00 8f 00 ca 74 00 00 00 03 04 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 00 00 01 04 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 02 ff 15 24 dd 04 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff e0 69 23
0001.040 Packet from SLIP of length 64 - write TUN
0000 60 02 b5 e1 00 18 3a ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 01 ff 4c 00 31 87 00 a9 3b 00 00 00 00 fd 00 78 78 78 78 00 00 02 80 e1 ff fe 4c 00 31
0001.041 Packet from SLIP of length 72 - write TUN
0000 6a 0b 05 db 00 20 00 01 fe 80 00 00 00 00 00 00 02 80 e1 ff fe 4c 00 31 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 02 3a 00 05 02 00 00 01 00 83 00 9d 24 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 02