contiki-os/contiki #1562: Some questions about ieee mode of cc1310

xioabudian, 2016-03-21:
Hello,

I just used this part of code,but the result of serial port is

'rf_radio_setup:CMD_RADIO_SETUP wait,CMDSTA=0x00000001,
status=0x0803'.

This error is caused by an error CMD_RADIO_SETUP command
parameters,but I can't know how to set right parameters.Would
you mind giving me some examples?
Kind regards.
following is the code
rf_radio_setup()
{
uint32_t cmd_status;
rfc_CMD_RADIO_SETUP_t cmd;

/* Create radio setup command */
rf_core_init_radio_op((rfc_radioOp_t *)&cmd, sizeof(cmd), CMD_RADIO_SETUP);

cmd.txPower.IB = tx_power_current->register_ib;
cmd.txPower.GC = tx_power_current->register_gc;
cmd.txPower.tempCoeff = tx_power_current->temp_coeff;
cmd.pRegOverride = ieee_overrides;
cmd.mode = 0x01;

/* Send Radio setup to RF Core */
if(rf_core_send_cmd((uint32_t)&cmd, &cmd_status) != RF_CORE_CMD_OK) {
PRINTF("rf_radio_setup: CMD_RADIO_SETUP, CMDSTA=0x%08lx, status=0x%04x\n",
cmd_status, cmd.status);
return RF_CORE_CMD_ERROR;
}

/* Wait until radio setup is done */
if(rf_core_wait_cmd_done(&cmd) != RF_CORE_CMD_OK) {
PRINTF("rf_radio_setup: CMD_RADIO_SETUP wait, CMDSTA=0x%08lx, status=0x%04x\n",
cmd_status, cmd.status);
return RF_CORE_CMD_ERROR;
}

return RF_CORE_CMD_OK;
}

g-oikonomou, 2016-03-22:
The CC1310 does not support IEEE mode. With CC1310, use PROP mode
George

xioabudian, 2016-03-22:
But CC13xx, CC26xx SimpleLinkTM Wireles MCU
Technical Reference Manual says it support  IEEE 802.15.4,
I can't understand that.
xiaobudian

g-oikonomou, 2016-03-22:
So there is what the TRM says and then there is what Contiki supports

Contiki supports:

* CC2650: IEEE mode (2.4GHz) and BLE mode. CC2630 and CC2640 _should_ also work ("should" is a keyword).
* CC1310: PROP mode (sub-ghz)

And just to be clear: I don't think the TRM suggests anywhere that the CC1310 can do 2.4GHz .15.4 :)

George

xioabudian, 2016-03-22:
Thank you very much!

xiaobudian

closed 2016-03-29 by alignan