contiki-os/contiki #1563: UART communication between collect view and UDP-sink crashed.

LanshunNie, 2016-03-22:
Hi, all,
RPL-collect and collect view is very good for tesing and observing the behavior of network. 
We use it a lot in our project. It works quite well most of the time. However, the UART communication between collect view and UDP-sink occasionally crashed and the UDP-Sink node could not push data to PC. I wonder what may cause this happen. Also, is SLIP/TUNSLIP still a reliable solution since it also rely on Serial Communication? Any help or advice on this will be appreciated. 
Thank you. 
Lanshun

alignan, 2016-03-25:
Could you please at least paste the error or try to provide more information?

LanshunNie, 2016-03-25:
Thanks alignan.
The most difficult part is we can not get any error information, : (
When it crashed, the terminal of Ubuntu does not refresh any more.
Then we attached the root node to another PC and tried to read something from it, nothing came out.
The happiness also seems to be random. 
So I am totally lost in finding the reason. We use a platform based on MSP430F5418A+ CC1120. 

Would you please just tell me whether tunslip and SLIP is widely used in practical WSN project and works reliably?
Thanks again.

alignan, 2016-03-28:
When it crashes, does the device has been assigned a new USB port?  for example if you used `/dev/ttyUSB0` it ends up assigned to `/dev/ttyUSB1` or alike?

You could always assign your device an `udev` rule so it always is assigned to a given port, and use applications like `supervisor` to re-launch the `tunslip6` script if crashes.  Google both alternatives.

LanshunNie, 2016-03-28:
Thanks alignan. 
I think this is quite helpful. Will check and try it.

alignan, 2016-03-29:
Great, thanks! If you happen to find any indication about what might went wrong, please re-open or make a PR!

closed 2016-03-29 by alignan