contiki-os/contiki #1567: How to develop blackhole, rank  and selective forwarding attack

Dmi10000, 2016-03-25:
Hello all,
Please I need help in developing the following attack types:
1. Rank attack
2. Blackhole attack
3. Selective forwarding attack
I will really appreciate your help.

Thank you.
Dave.

closed 2016-03-25 by alignan

KwameAsanteAdjei, 2020-08-24:
sad to find out that you've still not had a response after 4 years.
i really need to know how to develop SFA as well

miladkheirabi, 2024-08-26:
Lol. no response after 8 years. I am need to implement SFA.

hknaydin, 2024-12-17:
Hello,
For developing the following attack types: Rank Attack, Blackhole Attack, and Selective Forwarding Attack, you can refer to the following GitHub repository, which contains the necessary implementations and resources:

[https://github.com/dhondta/rpl-attacks/tree/master
](https://github.com/dhondta/rpl-attacks/tree/master
)
This repository provides a comprehensive setup for simulating these RPL (Routing Protocol for Low-Power and Lossy Networks) attacks, including relevant code, configurations, and instructions. It will be a good starting point for understanding and implementing these attacks.

I hope this helps!

miladkheirabi, 2025-02-16:
Hi,
I have the same issue. Currently, I am studying the Contiki NG documentation to understand the simulator. Would anyone happen to know how to make a log file from the simulation that works with run-analysis.py to generate charts?

far-voyager21, 2025-02-18:
Hi, I am an M.Tech student working on my dissertation, and I am currently
simulating RPL attacks (Blackhole, MITM, and DDoS) using the Cooja
simulator in Contiki. I am facing difficulties in implementing these
attacks, particularly in modifying the RPL protocol and writing the
necessary C code files.

I would greatly appreciate any guidance, sample code, or references that
could help in successfully simulating these attacks. If there are any
existing implementations or related discussions, please let me know. I came
across a github repo dhondta but I want the code files of these attacks so
that later on I can make some changes according to my project.

Looking forward to your support. Thanks in advance!