contiki-os/contiki #1568: error "Reading outside memory: 0xfa8b6"

htm1112, 2016-03-27:
I am trying to test some methods inside the IPsec and IKEv.2 protocol, using Contiki OS as implemented by the Cooja simulation environment.

I use Ubuntu 14.04 LTS 32-bit

gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.

msp430-gcc (GCC) 4.7.2 20120920 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.

I using Contiki-IPsec 2.7 OS, also I’m using Wismote in Cooja simulation 

but every time I try to run it, it always gives me the same error 
  
I'm very new in Contiki, if you can give me a small hint how I do it or any idea to overcome this issue is highly appreciated. 
Thanking you in advanced.

##########  java stack trace  ##########

se.sics.cooja.ContikiError
              at se.sics.cooja.mspmote.MspMote.execute(MspMote.java:324)
              at se.sics.cooja.mspmote.MspMote.execute(MspMote.java:291)
              at se.sics.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
              at se.sics.cooja.Simulation.run(Simulation.java:280)
              at java.lang.Thread.run(Thread.java:745)
Caused by: se.sics.mspsim.core.EmulationException: Reading outside memory: 0xfa8b6
              at se.sics.mspsim.core.MSP430Core$1.read(MSP430Core.java:162)
              at se.sics.mspsim.core.MSP430Core.emulateOP(MSP430Core.java:1003)
              at se.sics.mspsim.core.MSP430.stepMicros(MSP430.java:253)
              at se.sics.cooja.mspmote.MspMote.execute(MspMote.java:318)
              ... 4 more

              
              
##########  contiki error  ##########
Reading outside memory: 0xfa8b6
Stack Trace: number of calls: 24 PC: $fa8b6
  prf (sa.c) called from PC: $2191e (elapsed: 45283)
  prf_plus (sa.c) called from PC: $20056 (elapsed: 257530)
  ike_statem_get_ike_keymat (sa.c) called from PC: $2023c (elapsed: 51167089)
  ike_statem_parse_sa_init_msg (sa.c) called from PC: $1aa1c (elapsed: 51789229)
  ike_statem_state_parse_initreq (sa.c) called from PC: $1a540 (elapsed: 51849596)
  ike_statem_enterstate (sa.c) called from PC: $1a5a2 (elapsed: 51909682)
  ike_statem_setup_responder_session (sa.c) called from PC: $1a5e8 (elapsed: 52320397)
  ike_statem_incoming_data_handler (sa.c) called from PC: $1a30a (elapsed: 52387783)
  process_thread_ike2_service (local in ike.c) called from PC: $1280e (elapsed: 52415873)
  call_process (local in process.c) called from PC: $12aca (elapsed: 52415910)
  process_post_synch (sa.c) called from PC: $14a6c (elapsed: 52415923)
  tcpip_uipcall (sa.c) called from PC: $16e98 (elapsed: 52415966)
  uip_process (sa.c) called from PC: $149b6 (elapsed: 52833617)
  process_thread_tcpip_process (local in tcpip.c) called from PC: $1280e (elapsed: 52833660)
  call_process (local in process.c) called from PC: $12aca (elapsed: 52833697)
  process_post_synch (sa.c) called from PC: $1467a (elapsed: 52833710)
  tcpip_input (sa.c) called from PC: $14620 (elapsed: 52833719)
  input (local in sicslowpan.c) called from PC: $1237c (elapsed: 52836394)
  packet_input (local in nullmac.c) called from PC: $124a0 (elapsed: 52836401)
  packet_input (local in nullrdc.c) called from PC: $1877c (elapsed: 52839482)
  process_thread_cc2520_process (local in cc2520.c) called from PC: $1280e (elapsed: 52845593)
  call_process (local in process.c) called from PC: $128e4 (elapsed: 52845630)
  do_poll (local in process.c) called from PC: $12936 (elapsed: 52845706)
  process_run (sa.c) called from PC: $05f12 (elapsed: 52845723)

joakimeriksson, 2016-03-29:
reading outside of memory usually indicates that there is a bug in the code. And the code called seems to be pdf in sa.c. This is I would say >90% of the times a bug in the code - but in the worst case it could be a bug in the simulation / emulation platform. Have you tried running the same on the native platform without problems?

closed 2016-04-05 by nvt

sra1rdy, 2018-08-20:
@htm1112  did you get any solution?.. It would be great if u share the solution.

mhshd, 2019-03-02:
I have the same problem. would you please tell me how did you solve it?