contiki-os/contiki #1572: Contiki with multiple radios

davidsantosb, 2016-03-30:
I have recently ported contiki to my FWnode with the most basic functions.Right now I'm using the MCU MSP430F5438 and the radio CC2520.

But I have also two more radios (CC2500 and CC1101) on my node that I would like to use as wake up radios. Is there a way to define more than one radio on contiki network stack? or to use these radios simultaneously?

Thanks in advance

~ David

joakimeriksson, 2016-03-30:
I did a quick test with that and created a dual-radio driver so that both radios could be used. But it will require selection of radio on outgoing packets, etc. Not too hard to do a basic implementation of two-radios but to fully handle it is a bit trickier.

davidsantosb, 2016-03-30:
Thanks, @joakimeriksson . So should I include both radio drivers over the same driver?

I am really new at this but I guess I have to define NETSTACK_CONF_RADIO with the driver that will handle both of the radios, am I right?

joakimeriksson, 2016-03-30:
Yes, you basically create a virtual driver that is a “proxy” in between the radios and the upper layers. That is
at least how I did it. And add a few API’s for switching between then for outgoing packets
 (and/or add packet buf attributes).

I will actually work more with this next week and might be able to upstream some of the features of it
fairly soon (if I get an ok on that).

Best regards,
— Joakim

stebosisio, 2016-04-13:
Hello,

we are currently working at a "network level" multiradio solution for Contiki, we spoke some times ago about it on Contiki developers ML, collecting several positive feedbacks:

https://sourceforge.net/p/contiki/mailman/message/34919919/

I don't know if you have already read the above message, but, if not, we will really appreciate if you will give it a look and provide us some feedbacks. Obviously we are also, in general, very happy to exchange ideas about the "multiradio problems" and keep in touch with other people that are facing it. In any case as soon as our code will be "stable" we will publish it for community review and comments.

davidsantosb, 2016-04-21:
Hi @stebosisio,

Well it is very similar to what I've been trying to do. The thing is that I am porting a new node that we have and it has three nodes so basically we are trying to use one radio as principal radio and the other two radios as wake up radios so that we can save battery from the nodes to increase general lifetime of the overall network. I'm still working on the separate drivers and joining them together.

Still I am very interested on sharing ideas and I will also push the whole port of the nodes that we have been working on as soon as we get the "proxy" driver running. Please let me know how you are doing.

rajeev1986, 2016-06-13:
Hi @davidsantosb @stebosisio ,
Any latest updates on above "multi-stack" multi-radio solution  for Contiki and Cooja? Very interesting for my work as well, but if there is  some base work then it will make work faster for integrating and for improvements. 

Looking forward to any feedback on above.