contiki-os/contiki #1573: cc2650 sensortag not working with 6lbr anymore

janeksz, 2016-03-30:
now doesn't work at all
http://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/495080/1806406#1806406

laurentderu, 2016-03-31:
Have you tried changing the default 6LoWPAN context 0 from aaaa:: to 00fd:: in 6lbr ?

janeksz, 2016-04-01:
Thanks,
It works now (after changing aaaa:: to 00fd::), but why?

tonnenpinguin, 2016-04-01:
Because of pr #1297 getting merged!

closed 2016-04-01 by alignan

tonnenpinguin, 2016-04-14:
One thing though: Why do we have to set the prefix to 00fd rather than fd00?

joakimeriksson, 2016-04-19:
if you set to 00fd neither side will compress the prefix. But either aaaa:: or fd00:: will leave one of the sides compressing and the other not. So pick any arbitrary prefix except aaaa:: or fd00:: will probably work fine.