contiki-os/contiki #1578: Does 6LoWPAN stack support DHCPv6?

LanshunNie, 2016-04-01:
Hi, all,
Perhaps it is a stupid question. When we use 15.4 radio such as CC1120, it does not carry a unique MAC address. We have to burn a unique node ID to each mote in order to get IPv6 address by stateless auto-configuration. This is a time consuming work when there are a large number of motes. 
So I wonder whether the 6LoWPAN stack of Contiki 3.x supports DHCPv6. Can I rely on DHCP v6 to allocate IP v6 address? I did find some codes in directory /apps, but there is no documents and I have difficulty in figuring out the functions of these codes.
Thanks a lot.
Lanshun

vsaw, 2016-04-01:
The DHCP client right now is IPv4 as far as I remember. DHCPv6 is not very popular as stateless autoconfiguration essentially does the same thing without requiring additional infrastructure/servers.

That being said. You'll still need unique MAC addresses otherwise your neighbor table will be corrupted and the network will fail. If you don't have unique MAC Addresses you can try to randomly create one, then create a link local address from it and perform a duplicate address detection before claiming the MAC and IP addresses.

LanshunNie, 2016-04-02:
Got it.
Thanks vsaw.
I guess duplicate address detection is not easy and costly.

closed 2016-04-05 by nvt