contiki-os/contiki #1580: Unable to build atarixl target

rkujawa, 2016-04-02:
Hello. I am attempting to build atarixl target from MacOS X. I have prepared cc65 toolchain from master branch of their GitHub repository.

The build process stops with the following error:
```
make: *** No rule to make target `obj_atarixl/mtarch-asm.S', needed by `obj_atarixl/mtarch-asm.o'.  Stop.
```

It seems like it is unable to build the assembly source. The `ca65` assembler is in $PATH and works as expected.

Whole log:
```
$ make TARGET=atarixl
mkdir obj_atarixl
  CC        ../../apps/serial-shell/serial-shell.c
  CC        ../../apps/shell/shell.c
  CC        ../../apps/shell/shell-reboot.c
  CC        ../../apps/shell/shell-vars.c
../../apps/shell/shell-vars.c(80): Warning: Condition is always true
../../apps/shell/shell-vars.c(81): Warning: Condition is always true
  CC        ../../apps/shell/shell-ps.c
  CC        ../../apps/shell/shell-blink.c
  CC        ../../apps/shell/shell-text.c
  CC        ../../apps/shell/shell-time.c
  CC        ../../apps/shell/shell-file.c
  CC        ../../apps/shell/shell-run.c
  CC        ../../apps/shell/shell-coffee.c
  CC        ../../apps/shell/shell-power.c
  CC        ../../apps/shell/shell-base64.c
  CC        ../../apps/shell/shell-memdebug.c
  CC        ../../apps/shell/shell-powertrace.c
  CC        ../../apps/shell/shell-crc.c
  CC        ../../apps/shell/shell-rime.c
  CC        ../../apps/shell/shell-sendtest.c
  CC        ../../apps/shell/shell-rime-ping.c
  CC        ../../apps/shell/shell-rime-sniff.c
  CC        ../../apps/shell/shell-rime-netcmd.c
  CC        ../../apps/shell/shell-rime-debug.c
  CC        ../../apps/shell/shell-rime-debug-runicast.c
  CC        ../../apps/shell/shell-rime-sendcmd.c
  CC        ../../apps/shell/shell-download.c
  CC        ../../apps/shell/shell-rime-neighbors.c
  CC        ../../apps/shell/shell-rime-unicast.c
  CC        ../../apps/shell/shell-netperf.c
  CC        ../../apps/shell/shell-collect-view.c
  CC        ../../apps/shell/shell-wget.c
  CC        ../../apps/shell/shell-httpd.c
  CC        ../../apps/shell/shell-irc.c
  CC        ../../apps/shell/shell-tcpsend.c
  CC        ../../apps/shell/shell-udpsend.c
  CC        ../../apps/shell/shell-ping.c
  CC        ../../apps/shell/shell-netstat.c
../../apps/shell/shell-netstat.c(142): Warning: `file_name' is defined but never used
../../apps/shell/shell-netstat.c(142): Warning: `tcp_name' is defined but never used
../../apps/shell/shell-netstat.c(142): Warning: `proc_name' is defined but never used
  CC        ../../apps/collect-view/collect-view.c
  CC        ../../apps/collect-view/collect-view-template.c
  CC        ../../apps/webserver/http-strings.c
  CC        ../../core/net/ip/psock.c
  CC        ../../core/lib/memb.c
  CC        ../../apps/webserver/httpd-cfs.c
  CC        ../../apps/webserver/urlconv.c
  CC        ../../apps/webserver/webserver-nogui.c
  CC        ../../apps/webbrowser/webclient.c
  CC        ../../apps/webbrowser/http-user-agent-string.c
  CC        ../../apps/irc/ircc.c
  CC        ../../apps/irc/ircc-strings.c
  CC        ../../apps/telnet/telnet.c
  CC        ../../apps/powertrace/powertrace.c
  CC        ../../apps/shell/shell-dsc.c
  CC        ../../apps/webserver/webserver-dsc.c
  CC        ../../apps/webbrowser/www-dsc.c
  CC        ../../apps/irc/irc-dsc.c
  CC        ../../apps/telnet/telnet-dsc.c
  CC        ../../core/ctk/ctk-conio.c
  CC        ../../core/lib/petsciiconv.c
  CC        ../../core/cfs/cfs-posix-dir.c
  CC        ../../platform/atarixl/./contiki-main.c
  CC        ../../cpu/6502/lib/log.c
  CC        ../../platform/atarixl/lib/error.c
  CC        ../../cpu/6502/lib/unload.c
  CC        ../../cpu/6502/lib/config.c
  CC        ../../cpu/6502/ctk/ctk-mouse.c
  CC        ../../cpu/6502/sys/clock.c
  CC        ../../cpu/6502/sys/mtarch.c
make: *** No rule to make target `obj_atarixl/mtarch-asm.S', needed by `obj_atarixl/mtarch-asm.o'.  Stop.
```

oliverschmidt, 2016-04-02:
As you can see i.e. from https://travis-ci.org/contiki-os/contiki/jobs/119999409 the ATARIXL build works generally just fine. Something is broken in your local setup.

If there would be an issue with your `ca65` then the error message would definitively differ from `*** No rule to make target`. That message means that there's in issue with make and the interpretation of the Makefiles.

closed 2016-04-02 by oliverschmidt

rkujawa, 2016-04-02:
@oliverschmidt Thanks for your reply.

I suspect this was caused by too old `make` version. XCode ships version 3.81 by default. I have installed version 4.1, using it allowed me to proceed further with the build.