contiki-os/contiki #1581: Get error "Could not open flash to save config" when test cc26xx-web-demo on CC2650STK

chenek, 2016-04-04:
Get error "Could not open flash to save config" when test cc26xx-web-demo on CC2650STK. Anyone have idea how to fix it?

g-oikonomou, 2016-04-04:
How are you connecting to the tag's UART to see those messages? Debugger devpack or are you using the JTAG trick?

chenek, 2016-04-04:
I use Debugger devpack.

g-oikonomou, 2016-04-04:
OK I'll have a look at it more closely, I must say I've not been using devpacks a whole lot. What revision of the devpack are you using?

chenek, 2016-04-04:
It's rev: 1.2.0. The followings are my log output from cc26xx-we-demo.
Starting Contiki-3.x-2243-g5205198
With DriverLib v0.44336
TI CC2650 SensorTag
 Net: sicslowpan
 MAC: CSMA
 RDC: ContikiMAC, Channel Check Interval: 16 ticks
 RF: Channel 25
CC26XX Web Demo Process
CC26XX Web Server
CC26XX CoAP Server
6LBR Client Process
CC26XX MQTT Client Process
CC26XX Net UART Process
Could not open flash to load config

arurke, 2016-04-04:
Been a while since I used the devpack and I cannot remember having this problem, but in my notes I have the following link: http://forum.43oh.com/topic/8822-warning-jtag-interferes-with-sensortag-external-flash-access/ Does it match your situation? Also I remember I had to power cycle the sensortag after flashing new sw to a tag, or else the flash (and watchdog_reboot()) wouldn't work.

chenek, 2016-04-04:
I had tried to power recycle CC2650STK but it still doesn't work.

g-oikonomou, 2016-04-04:
Andreas, I too was suspecting that the devpack JTAG connection might be messing up with the SPI clock (same as what happens when programming over SmartRF Flash Prog, only the workaround there is a simple powercycle). This why I was talking about powercycling, here as well as in #1458.

This external link is very useful and explains it. I think we should probably update the platform's README about this one.

@chenek did you try the sequence from that forum post?

chenek, 2016-04-04:
Thanks George and Andreas,
After I close Flash Programmer 2, it works now.

g-oikonomou, 2016-04-04:
#1581
and
http://forum.43oh.com/topic/8822-warning-jtag-interferes-with-sensortag-external-flash-access/

closed 2016-04-04 by g-oikonomou